Kinovarobotics/ros2_kortex #151: Initial ROS 2 Release

moriarty, 2023-07-19:
I have bloomed Humble and Iron.

- [ ] https://github.com/ros/rosdistro/pull/38028
- [ ] https://github.com/ros/rosdistro/pull/38029

Rolling is held up by dependencies which are held for sync
- [ ] https://github.com/PickNikRobotics/ros2_robotiq_gripper/issues/33

moriarty, 2023-07-19:
After the the release for Humble, Iron and Rolling, I can clean up the todos in the README.md so that the default documentation is to use the package from the binary release

moriarty, 2023-07-27:
After testing initial release release I have made fixes and released again:

- [ ] https://github.com/ros/rosdistro/pull/38074
- [ ] https://github.com/ros/rosdistro/pull/38075
- [ ] https://github.com/ros/rosdistro/pull/38076

aalmrad, 2024-10-08:
Closing due to inactivity - reply to the thread if you need the issue to be reopened

closed 2024-10-08 by aalmrad