Kinovarobotics/ros2_kortex #158: Tracking Ticket: Arm64 Support

moriarty, 2023-07-27:
https://artifactory.kinovaapps.com/ui/native/generic-public/kortex/API/

Opening this to track that there is no Arm64 Support but also marking it as **wontfix** and **enhancement**

Currently there is no linux_arm64 build for Ubuntu:

[linux_armv7_artik710_gcc.zip](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/linux_armv7_artik710_gcc.zip)
[linux_armv7_imx6_gcc.zip](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/linux_armv7_imx6_gcc.zip)
[linux_armv7_jetson_gcc.zip](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/linux_armv7_jetson_gcc.zip)
[linux_x86-64_x86_gcc.zip](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/linux_x86-64_x86_gcc.zip)
[protos.tar.gz](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/protos.tar.gz)
[windows_x86-64_mingw.zip](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/windows_x86-64_mingw.zip)
[windows_x86-64_msvc-2015.zip](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/windows_x86-64_msvc-2015.zip)
[windows_x86-64_msvc-2017.zip](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/windows_x86-64_msvc-2017.zip)
[windows_x86-64_msvc-2019.zip](https://artifactory.kinovaapps.com/artifactory/generic-public/kortex/API/2.6.0/windows_x86-64_msvc-2019.zip)

moriarty, 2023-07-27:
This issue is more for documentation and linking to the ros2_buildfarm_config.
If the Kortex Api were made available on arm64 then it could be re-enabled.

- [ ] https://github.com/ros2/ros_buildfarm_config/pull/280
- [ ] https://github.com/ros2/ros_buildfarm_config/pull/279
- [ ] https://github.com/ros2/ros_buildfarm_config/pull/278

moriarty, 2023-08-03:
Blacklist PRs have been merged. Closing this issue.

@felixmaisonneuve if Arm64 support is ever available you'll need to undo the blacklist from ros2/ros_buildfarm_config

closed 2023-08-03 by moriarty

felixmaisonneuve, 2023-08-03:
@moriarty Got it, thank you

ssapsu, 2024-10-06:
> https://artifactory.kinovaapps.com/ui/native/generic-public/kortex/API/
>
> Opening this to track that there is no Arm64 Support but also marking it as **wontfix** and **enhancement**
>
> Currently there [...]

[image]
There's a arm aarch64. So then can we build it in arm64?

ssapsu, 2024-10-07:
Hi guys I managed to do it!!

https://github.com/user-attachments/assets/4367b4c0-a76d-41b0-a4c4-59add0a17bf9