Kinovarobotics/ros2_kortex #169: Linux architecture issue

gtirth, 2023-08-31:
I am trying to run ros2_kortex inside a docker on stereolab's zedbox orin nx. But when I tried to compile the sources i got an error that aarch64 is not supported. 
[image]

Is there a workaround for this?

moriarty, 2023-08-31:
No, at this time Kinova does not provide an arm64 kortex api

gtirth, 2023-08-31:
> No, at this time Kinova does not provide an arm64 kortex api

Would it be possible to use moveit?

moriarty, 2023-08-31:
Duplicate of #158

Arron-Indrotek, 2023-09-12:
Hi @moriarty, We where hoping you could help us get the Kinova gen3 ROS 2 drivers working on NIVIDA Jetson. Are you the best person to collaborate with? Do you have the source secure to compile the stack on the ARM cortex?... or is there someone else we should colab with?

moriarty, 2023-09-13:
Which Jetson? The older Jetson with older arm chip might be possible but not the newer models. I don't have access to the kortex api source code but have asked Kinova for newer arm chip support.

ffournier-kinova, 2023-09-13:
Kinova supports the kortex_api package on ARM (Jetson v7 compiled with gcc) but it is not currently added as a supported architecture in the ros2_kortex_api package. The architecture would have to be added to the CMake (https://github.com/Kinovarobotics/ros2_kortex/blob/main/kortex_api/CMakeLists.txt#L23) and link to the aarch build of kortex API (https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_armv7_jetson_gcc.zip). you should then be able to compile following the instructions in the README (https://github.com/Kinovarobotics/ros2_kortex#protobuf)

moriarty, 2023-09-14:
Yes I don't have an armv7 board myself setup to test it out right now, and I thought the latest Jetsons were armv8 but if the kortex_api is available this should be possible.

We had originally discussed arm64 support as part of the ROS 2 drivers project because it's a common with all the new MacBook chips.

As also discussed it would be really great if enough of the underlying kortex api could be open sourced so that as new chips are released support could be added with less effort and easier to support and debug.

I'm currently pretty busy and no longer have a Kinova Gen3 to test on hardware, but I think I have an old Jetson in storage that I could test on...

If the binary api is available then it should be just a bit of CMake to pull that in based on the architecture.

autonomousTurtle, 2023-12-15:
I'm working on a gen3 with nvidia orin dev kit with jetpack 6 (Ubuntu 22). Using the suggestion above the kortex api builds, but the kortex driver does not - even with the same change in the CMake. 
[image]
Happy to help test and debug as I work through this.

autonomousTurtle, 2024-03-12:
I was able to get my system up and running, but will not claim that this is scalable or a best practice, merely a workaround. I first changed to the link and architecture type in cmake as moriarty suggested, but the extract feature through fetch content was expecting a tar file, not a zip. Threw up build errors, and I'm not familiar enough with fetch content for my 2 minutes of googling to produce a solve. Therefore, here is my working workaround: 

_this in on Nvidia Orin Dev Kit running ROS2 Humble_

1. I downloaded the files from the kortex API (https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_armv7_jetson_gcc.zip) and extracted into a folder in my ros2 workspace. 
2. I pointed the scripts to that downloaded and extracted folder. 

Working cmake code for kortex_api package below: 

```
cmake_minimum_required(VERSION 3.14)
#include(FetchContent)
#include(ExternalProject)
#FetchContent_Declare(
#  kinova_binary_api
#  URL https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_armv7_jetson_gcc.zip
#  URL_HASH MD5=988d210bcff2009a96659c115bb4fc62
#)

#FetchContent_MakeAvailable(kinova_binary_api)
#set(FETCHCONTENT_ZIP_USE_SYSTEM ON)

set(KINOVA_BINARY_API_DIR "/home/apricity/workspace/ros2_kortex_ws/linux_aarch64_gcc_7.4")

project(kortex_api)

#Future:
# convert API_URL components from CMAKE_SYSTEM variables.
# The kinova artifactory server artifacts have slightly different expectations and they need some transformations
# Could build something like: (URL_OS)_(URL_PROCESSOR)_(?)_(URL_COMPILER).zip
# string(REPLACE "_" "-" URL_PROCESSOR ${CMAKE_SYSTEM_PROCESSOR}) # x86_64 -> x86-64
# string(TOLOWER ${CMAKE_SYSTEM_NAME} URL_OS) # Linux -> linux

#Current: only support Linux x86_64
#if(CMAKE_SYSTEM_NAME STREQUAL "Linux" AND CMAKE_SYSTEM_PROCESSOR STREQUAL "x86_64")
#  set(API_URL https://artifactory.kinovaapps.com:443/artifactory/generic-public/kortex/API/2.5.0/linux_x86-64_x86_gcc.zip)
#elseif(CMAKE_SYSTEM_NAME STREQUAL "Linux" AND CMAKE_SYSTEM_PROCESSOR STREQUAL "aarch64")
#  set(API_URL https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_armv7_jetson_gcc.zip)
#else()
#  set(API_URL "")
#  message(FATAL_ERROR "Unsupported System: currently support is for Linux x68_64. Detected ${CMAKE_SYSTEM_NAME} and ${CMAKE_SYSTEM_PROCESSOR}")
#endif()

#ExternalProject_Add(kinova_binary_api
#  URL ${API_URL}
#  CONFIGURE_COMMAND ""
#  BUILD_COMMAND ""
#  INSTALL_COMMAND ""
#)

find_package(ament_cmake REQUIRED)

ament_export_include_directories(include/kortex_api)

install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/client/ DESTINATION include/kortex_api)
install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/common/ DESTINATION include/kortex_api)
install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/messages/ DESTINATION include/kortex_api)
install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/client_stubs/ DESTINATION include/kortex_api)
install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/google/protobuf DESTINATION include/kortex_api/google)

ament_package()
```

You will also need to update the kortex driver cmake in the same way. So far no issues, I will report back if something else is broken with this process in use that I haven't discovered yet. 

I also had to delete all references to issac sim for the package to work, but I believe that error was unrelated.

Shiven110504, 2024-06-13:
Hi, how would you change the kortex driver cmake. Could you point in what we will need to change.

Jianliang-Shen, 2025-04-25:
> I was able to get my system up and running, but will not claim that this is scalable or a best practice, merely a workaround. I first changed to the link and architecture type in cmake as @ffournier-kinova [suggested](https://github.com/Kinovarobotics/ros2_kortex/issues/169#issuecomment-1718040052), but the extract feature through fetch content was expecting a tar file, not a zip. Threw up build errors, and I'm not familiar enough with fetch content for my 2 minutes of googling to produce a solve. Therefore, here is my working workaround:
> 
> _this in on Nvidia Orin Dev Kit running ROS2 Humble_
> 
> 1. I downloaded the files from the kortex API (https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_armv7_jetson_gcc.zip) and extracted into a folder in my ros2 workspace.
> 2. I pointed the scripts to that downloaded and extracted folder.
> 
> Working cmake code for kortex_api package below:
> 
> ```
> cmake_minimum_required(VERSION 3.14)
> #include(FetchContent)
> #include(ExternalProject)
> #FetchContent_Declare(
> #  kinova_binary_api
> #  URL https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_armv7_jetson_gcc.zip
> #  URL_HASH MD5=988d210bcff2009a96659c115bb4fc62
> #)
> 
> #FetchContent_MakeAvailable(kinova_binary_api)
> #set(FETCHCONTENT_ZIP_USE_SYSTEM ON)
> 
> set(KINOVA_BINARY_API_DIR "/home/apricity/workspace/ros2_kortex_ws/linux_aarch64_gcc_7.4")
> 
> project(kortex_api)
> 
> #Future:
> # convert API_URL components from CMAKE_SYSTEM variables.
> # The kinova artifactory server artifacts have slightly different expectations and they need some transformations
> # Could build something like: (URL_OS)_(URL_PROCESSOR)_(?)_(URL_COMPILER).zip
> # string(REPLACE "_" "-" URL_PROCESSOR ${CMAKE_SYSTEM_PROCESSOR}) # x86_64 -> x86-64
> # string(TOLOWER ${CMAKE_SYSTEM_NAME} URL_OS) # Linux -> linux
> 
> #Current: only support Linux x86_64
> #if(CMAKE_SYSTEM_NAME STREQUAL "Linux" AND CMAKE_SYSTEM_PROCESSOR STREQUAL "x86_64")
> #  set(API_URL https://artifactory.kinovaapps.com:443/artifactory/generic-public/kortex/API/2.5.0/linux_x86-64_x86_gcc.zip)
> #elseif(CMAKE_SYSTEM_NAME STREQUAL "Linux" AND CMAKE_SYSTEM_PROCESSOR STREQUAL "aarch64")
> #  set(API_URL https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_armv7_jetson_gcc.zip)
> #else()
> #  set(API_URL "")
> #  message(FATAL_ERROR "Unsupported System: currently support is for Linux x68_64. Detected ${CMAKE_SYSTEM_NAME} and ${CMAKE_SYSTEM_PROCESSOR}")
> #endif()
> 
> #ExternalProject_Add(kinova_binary_api
> #  URL ${API_URL}
> #  CONFIGURE_COMMAND ""
> #  BUILD_COMMAND ""
> #  INSTALL_COMMAND ""
> #)
> 
> find_package(ament_cmake REQUIRED)
> 
> ament_export_include_directories(include/kortex_api)
> 
> install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/client/ DESTINATION include/kortex_api)
> install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/common/ DESTINATION include/kortex_api)
> install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/messages/ DESTINATION include/kortex_api)
> install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/client_stubs/ DESTINATION include/kortex_api)
> install(DIRECTORY ${KINOVA_BINARY_API_DIR}/include/google/protobuf DESTINATION include/kortex_api/google)
> 
> ament_package()
> ```
> 
> You will also need to update the kortex driver cmake in the same way. So far no issues, I will report back if something else is broken with this process in use that I haven't discovered yet.
> 
> I also had to delete all references to issac sim for the package to work, but I believe that error was unrelated.

I used https://artifactory.kinovaapps.com:443/artifactory/generic-public/kortex/API/2.6.0/linux_aarch64_gcc_7.4.zip to build kinova in Ubuntu 24.04 Virtual Machine with my Mac M2 silicone. It works! Let me try after the building.

closed 2026-02-04 by aalmrad