Kinovarobotics/ros2_kortex #172: kortex_moveit.launch.py

gtirth, 2023-09-13:
Hello,

I am trying to use kortex_moveit for multiple gen3 arm but running into an issue when i try to launch it.

[image]

Can someone please suggest a solution for this.

moriarty, 2023-09-13:
This seems familiar but I'm not on my computer right now.

Which version of MoveIt are you using?

Can you provide your launch file?

gtirth, 2023-09-13:
file:///home/tirth/launch_file.txt

I am using moveit_2 for humble distro.

I have attached file that i created for custom robot that has 2 gen3 arms. I was able to create the move_it package with moveit_setup_assistant but I have not been able to create a launch file that would allow me move the arms. I was able to test and move individual arms with kinova_gen3_7dof_robotiq_2f_85_moveit_config. 

I would appreciate if you help or redirect me to some resources which would allow to do that

gtirth, 2023-09-14:
Hello,

I am using moveit_2 for humble distro. I have attached file that i created for custom robot that has 2 gen3 arms. There might be some more changes which i would have to do but i was not able to launch it so i did not do all the modifications in the launch file. I was able to create the move_it package with moveit_setup_assistant but I have not been able to create a launch file that would allow me move the arms. I was able to test and move individual arms with kinova_gen3_7dof_robotiq_2f_85_moveit_config.

I would appreciate if you help or redirect me to some resources which would allow to do that

[cid:00f2f77a-b188-4121-ace3-e53504f4b94f]
Tirth Gajera
Technologist
InDro Robotics Inc.
(+1) 613 501 0646
***@***.***
www.indrorobotics.com<http://www.indrorobotics.com>

Caution: This email originated from outside of the organization. Please take care when clicking links or opening attachments, and never send money to anyone through an email or reply to an email asking to send money, it is a scam. When in doubt, contact your IT Department

# Copyright (c) 2021 PickNik, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# Author: Denis Stogl

import os

import yaml
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.conditions import IfCondition
from launch.substitutions import (
    Command,
    FindExecutable,
    LaunchConfiguration,
    PathJoinSubstitution,
)
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

def load_yaml(package_name, file_path):
    package_path = get_package_share_directory(package_name)
    absolute_file_path = os.path.join(package_path, file_path)

    try:
        with open(absolute_file_path) as file:
            return yaml.safe_load(file)
    except OSError:  # parent of IOError, OSError *and* WindowsError where available
        return None

def generate_launch_description():
    declared_arguments = []
    # Robot specific arguments
    declared_arguments.append(
        DeclareLaunchArgument(
            "robot_type", description="Type/series of robot.", choices=["gen3", "gen3_lite"], default_value="gen3"
        )
    )
    declared_arguments.append(DeclareLaunchArgument("dof", description="DoF of robot."))
    declared_arguments.append(
        DeclareLaunchArgument(
            "robot_ip", description="IP address by which the robot can be reached.", default_value= "192.168.42.191"
[... 213 lines cut ...]
    # rviz with moveit configuration
    rviz_config_file = PathJoinSubstitution(
        [FindPackageShare(moveit_config_package), "rviz", "moveit.rviz"]
    )
    rviz_node = Node(
        package="rviz2",
        condition=IfCondition(launch_rviz),
        executable="rviz2",
        name="rviz2_moveit",
        output="log",
        arguments=["-d", rviz_config_file],
        parameters=[
            robot_description,
            robot_description_semantic,
            ompl_planning_pipeline_config,
            robot_description_kinematics,
        ],
    )

    nodes_to_start = [
        move_group_node,
        rviz_node,
    ]

    return LaunchDescription(declared_arguments + nodes_to_start)

jalapatip, 2024-01-09:
@gtirth, I'm working on a multiarm setup too but haven't had much luck. I was wondering if you were able to set it up.