Kinovarobotics/ros2_kortex #173: replace() argument 2 must be str, not LaunchConfiguration

patrickwasp, 2023-09-15:
When trying to launch moveit on humble I get the following error

```
ros2 launch kinova_gen3_6dof_robotiq_2f_85_moveit_config robot.launch.py robot_ip:=192.168.8.20 --debug
```

```
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-09-15-01-25-24-736017-8bf77abc1655-207
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fdea4359840>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fdea4359840>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7fdea450f430>'
[DEBUG] [launch]: An exception was raised in an async action/event
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 336, in run_async
    raise completed_tasks_exceptions[0]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 230, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 250, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "/opt/ros/humble/share/kinova_gen3_6dof_robotiq_2f_85_moveit_config/launch/robot.launch.py", line 53, in launch_setup
    MoveItConfigsBuilder("gen3", package_name="kinova_gen3_6dof_robotiq_2f_85_moveit_config")
  File "/opt/ros/humble/lib/python3.10/site-packages/moveit_configs_utils/moveit_configs_builder.py", line 213, in robot_description
    self.__robot_description: load_xacro(
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_param_builder/utils.py", line 82, in load_xacro
    file = xacro.process_file(file_path, mappings=mappings)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1098, in process_file
    process_doc(doc, **kwargs)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1050, in process_doc
    eval_all(doc.documentElement, macros, symbols)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 977, in eval_all
    elif handle_macro_call(node, macros, symbols):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 790, in handle_macro_call
    scoped_symbols._setitem(name, eval_text(value, symbols), unevaluated=False)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 706, in eval_text
    results.append(handle_extension(lex.next()[1][2:-1]))
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 696, in handle_extension
    return eval_extension("$(%s)" % eval_text(s, symbols))
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 289, in eval_extension
    return resolve_args(s, context=substitution_args_context)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/substitution_args.py", line 322, in resolve_args
    resolved = _resolve_args(arg_str, context, commands)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/substitution_args.py", line 338, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/substitution_args.py", line 182, in _arg
    return resolved.replace('$(%s)' % a, _eval_arg(name=args[0], args=context['arg']))
TypeError: replace() argument 2 must be str, not LaunchConfiguration

[ERROR] [launch]: Caught exception in launch (see debug for traceback): replace() argument 2 must be str, not LaunchConfiguration
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fdea450f370>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fdea450f370>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7fdea45fcc40>'
```

this is my dockerfile

```
FROM osrf/ros:humble-desktop-full

RUN apt-get update \
    && apt-get install --yes \
    wget \
    ros-$ROS_DISTRO-moveit \
    ros-$ROS_DISTRO-cv-bridge \
    ros-$ROS_DISTRO-kortex-bringup \
    ros-$ROS_DISTRO-kinova-gen3-6dof-robotiq-2f-85-moveit-config \
    ros-$ROS_DISTRO-rmw-cyclonedds-cpp \
    python3-pip \
    && rm -rf /var/lib/apt/lists/*
```

felixmaisonneuve, 2023-09-27:
Hi @patrickwasp,

Have you tried running the command with the option `use_fake_hardware:=true`?
https://github.com/Kinovarobotics/ros2_kortex#usage

Best,
Felix

ssapsu, 2024-02-04:
@patrickwasp 

In my case, ros-distro: rolling was working

moriarty, 2024-02-04:
This was likely tested with:

ROS Humble + MoveIt installed from source off of main.

The MoveIt Project maintains the main branch compatible from source with Humble, Iron and Rolling... but doesn't or tries not to release breaking changes back to the ROS Humble released versions of MoveIt.

I suspect this error is ROS Humble + MoveIt installed via the ROS Humble repository. Check the MoveIt documentation and tutorials there is instructions on how to use older versions of ROS with the latest version of MoveIt.

aalmrad, 2024-10-07:
Closing due to inactivity - reply to the thread if you need the issue to be reopened

closed 2024-10-07 by aalmrad