Kinovarobotics/ros2_kortex #175: gen3_macro.xacro can't be used to generate urdf

patrickwasp, 2023-09-26:
running `xacro src/ros2_kortex/kortex_description/arms/gen3/6dof/urdf/gen3_macro.xacro` returns

```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from src/ros2_kortex/kortex_description/arms/gen3/6dof/urdf/gen3_macro.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="gen3_arm">
</robot>
````

this causes moveit setup assistant to crash with `Error:   No link elements found in urdf file`

felixmaisonneuve, 2023-09-27:
Hi @patrickwasp,

We usually run the xacro command on the gen3_robotiq_2f_85.xacro and gen3_robotiq_2f_140.xacro files (in kortex_description/robots folder), so some arguments are specified
https://github.com/Kinovarobotics/ros2_kortex/blob/a2f018ab8c39e97fe91f57c27b8e59852b029da4/kortex_description/robots/gen3_robotiq_2f_85.xacro#L6-L13

If you run the xacro command on the gen3_macro.xacro file directly, arguments are not specified, so things might not work. 
E.g. here the file cannot be found correctly because `dof` is not defined
https://github.com/Kinovarobotics/ros2_kortex/blob/main/kortex_description/arms/gen3/6dof/urdf/gen3_macro.xacro#L35

This probably explains why your file is empty.

What you could do is run the xacro command on the gen3.xacro file (in kortex_description/robots) and specify the arguments manually instead
something like `xacro src/ros2_kortex/kortex_description/robots/gen3.xacro arm:=gen3 dof:=6 vision:=false...`

Hope this helps,
Felix

patrickwasp, 2023-09-29:
Since this xacro file is in `kortex_description/arms/gen3/6dof/`, which already specifies 6dof, could we have 6 be set as the default in the gen3_macro.xacro file?

felixmaisonneuve, 2023-10-02:
You can do make the modification on your side, but since the default use case does not require you to convert this file directly, I would prefer not to change it

aalmrad, 2024-10-07:
Closing due du inactivity - reply to the thread if you need the issue to be reopened

closed 2024-10-07 by aalmrad