Kinovarobotics/ros2_kortex #178: view_robot.launch.py does not update to joint states

patrickwasp, 2023-09-30:
ros version: humble

running `ros2 launch kortex_description view_robot.launch.py` and pressing on "Randomize" in the "Joint State Publisher" window does not reliably update the arm positions. The terminal will say

```
[joint_state_publisher_gui-3] [INFO] [1696086847.268803477] [joint_state_publisher_gui]: Randomizing
```

but the robot remains upright. Quitting (<kbd>ctrl</kbd>+<kbd>c</kbd>) and rerunning `ros2 launch kortex_description view_robot.launch.py` occasionally works. I cannot reliably replicate when it works or doesn't.

felixmaisonneuve, 2023-10-02:
Could it be possible the randomization tries to move the arm in a configuration outside its joint limits?

patrickwasp, 2023-10-03:
No, that was just an example but trying to move any joint results in no motion.

aalmrad, 2024-10-08:
Hello, apologies for the late reply. We are recently trying to tackle the github issues on our repositories more actively. I could not replicate this issue on my end. I realize that many modifications were applied to the repository since then so please let me know in case you are still facing this issue the up-to-date main branch and we can work towards a solution.

Best,
Abed

yuvansharma, 2024-11-20:
I am also facing the same issue, both randomizing and manually setting values seems to have no effect.