Kinovarobotics/ros2_kortex #186: Add Tool Wrench Measurements to ROS2 Hardware Interface

eholum, 2023-10-30:
We would like to get started using force-feedback for control of the gen3 arms, but require access to tool force/torque measurements from the Kortex API. 

It looks like the API provides this information in the `tool_external_wrench_*` objects in the feedback message. However, just testing adding those values to the drivers [read()](https://github.com/Kinovarobotics/ros2_kortex/blob/a2f018ab8c39e97fe91f57c27b8e59852b029da4/kortex_driver/src/hardware_interface.cpp#L745) function does not seem to update the measured values.

```
  auto fx = feedback_.base().tool_external_wrench_force_x();
  auto fy = feedback_.base().tool_external_wrench_force_y();
  auto fz = feedback_.base().tool_external_wrench_force_z();
  auto tx = feedback_.base().tool_external_wrench_torque_x();
  auto ty = feedback_.base().tool_external_wrench_torque_y();
  auto tz = feedback_.base().tool_external_wrench_torque_z();
```

Given the conversation in https://github.com/Kinovarobotics/kortex/issues/52, it sounds like the wrench measurements are not available in low level control mode. However, even when switching the high level mode (and using the built in twist controller) we do not seem to get any information out of the API.

I have two questions,

1) Is it possible to expose these values through the Kortex API?
2) Are there any other built in functions we could use to compute the wrench or should we look into doing it on the fly with the KDL?

smoya23, 2024-04-11:
Hi @eholum,

1. You should be able to get the tool wrench values through the Kortex API high level control. You should however precede your getter calls by a call to the RefreshFeedBack() function of Kortex BaseCyclic client, as done in our example [here](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/102-High_level_movement/01-move_angular_and_cartesian.cpp#L189). Could you share a snippet of your code so that I can analyse if there is an error?

2. A part from the feedback of each actuator, there is nothing else.

Best,
Santiago

eholum, 2024-04-22:
Thanks for the response, @smoya23! We ended up using an external, wrist-mounted FTS and kind of put the internal calculation on hold. I think we can drop this.

closed 2024-04-22 by eholum