Kinovarobotics/ros2_kortex #188: does ros2_kortex works in windows 10?

Vasanth28897, 2023-11-18:
Hi, I installed ros2 iron in windows 10. I wanted to check whether the packages like kinova kortex(simulation) is working fine or not. I followed the instructions [here](https://docs.ros.org/en/crystal/Installation/Windows-Install-Binary.html) to install ros2 in windows, and [this](https://github.com/Kinovarobotics/ros2_kortex/blob/main/README.md) instructions to clone and build the package. After the rosdep installed, this command `rosdep install --ignore-src --from-paths src -y -r` gives the error like this
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros2_control_demo_example_7: Unsupported OS [windows]
hardware_interface: Unsupported OS [windows]
ros2_control_demo_example_2: Unsupported OS [windows]
ros2controlcli: Unsupported OS [windows]
kinova_gen3_7dof_robotiq_2f_85_moveit_config: Unsupported OS [windows]
ros2_control_demo_example_14: Unsupported OS [windows]
ros_gz_bridge: Unsupported OS [windows]
angles: Unsupported OS [windows]
ros2_control_demo_example_8: Unsupported OS [windows]
gz_ros2_control_tests: Unsupported OS [windows]
kortex_bringup: Unsupported OS [windows]
ros2_control_demo_example_10: Unsupported OS [windows]
ros2_control_demo_example_3: Unsupported OS [windows]
ros2_control_demo_example_5: Unsupported OS [windows]
ros2_control_demo_description: Unsupported OS [windows]
controller_manager: Unsupported OS [windows]
gazebo_ros2_control: Unsupported OS [windows]
ros_gz_image: Unsupported OS [windows]
gz_ros2_control: Unsupported OS [windows]
ros2_control_demo_example_4: Unsupported OS [windows]
ros2_control_demo_example_9: Unsupported OS [windows]
kinova_gen3_6dof_robotiq_2f_85_moveit_config: Unsupported OS [windows]
gz_ros2_control_demos: Unsupported OS [windows]
gazebo_ros2_control_demos: Unsupported OS [windows]
kortex_description: Unsupported OS [windows]
ros2_control_demo_example_12: Unsupported OS [windows]
ros2_control_demo_example_6: Unsupported OS [windows]
ros_gz_sim_demos: Cannot locate rosdep definition for [gz-sim7]
ros2_control_demo_example_1: Unsupported OS [windows]
ros_gz_sim: Unsupported OS [windows]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
and this `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release` gives the error like this

```
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Starting >>> ros2_control_demo_description
Starting >>> angles
Starting >>> ros_gz_interfaces
Starting >>> ros_gz_sim
Starting >>> kortex_description
Starting >>> joint_limits
Starting >>> kortex_api
Finished <<< ros2_control_demo_description [1.51s]
Finished <<< ros2_control_test_assets [1.83s]
--- stderr: kortex_api
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.27/Modules/FetchContent.cmake:1316 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  CMakeLists.txt:4 (FetchContent_Declare)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:28 (message):
  Unsupported System: currently support is for Linux x68_64.  Detected
  Windows and AMD64

---
Failed   <<< kortex_api [0.97s, exited with code 1]
Aborted  <<< kortex_description [1.24s]
Aborted  <<< angles [1.69s]
Aborted  <<< ros_gz_sim [1.83s]
Aborted  <<< controller_manager_msgs [4.27s]
Aborted  <<< ros_gz_interfaces [4.59s]
Aborted  <<< joint_limits [7.59s]

Summary: 2 packages finished [9.78s]
  1 package failed: kortex_api
  6 packages aborted: angles controller_manager_msgs joint_limits kortex_description ros_gz_interfaces ros_gz_sim
  5 packages had stderr output: controller_manager_msgs joint_limits kortex_api ros_gz_interfaces ros_gz_sim
  33 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)
```
we can see the CMake Error showing `Unsupported System: currently support is for Linux x68_64.  Detected Windows and AMD64`.  does this package really works in windows, or is it still in development? If it works fine, what are the steps i have to follow to make it work?

Vasanth28897, 2023-11-20:
So, if i add a check in the CMakelists.txt(in the kortex_api) file for windows instead of linux, and give the exact windows binary (windows_x86-64_msvc-2019.zip) url, will it work?

smoya23, 2024-04-11:
Hi @Vasanth28897,
As @moriarty mentionned, Windows support was not tested nor in the scope for the ros2_kortex repo. You might be able to make it work, but unfortunately it's something we cannot offer support for.

Best,
Santiago

closed 2024-07-26 by martinleroux