Kinovarobotics/ros2_kortex #202: How can use effort controller in ros2_kortex (ActuatorConfig)

odj3460, 2024-01-10:
We are modifying hardware_interface based on the API torque-control example.
I want to be able to give torque input at low-level, i.e. switch to the ros-controller's effort controller.
As of now, I have set it that way, but the following error occurs.

**error: no matching function for call to ‘Kinova::Api::ActuatorConfig::ActuatorConfigClient::ActuatorConfigClient()’**

In the basic hardware_interface.cpp, it was stated that ros2_kortex is not provided for effort input.
But even if I modify the source file, is it currently not possible?

I would like to use a controller other than the default joint-trajectory-controller and twist-controller.

To be precise, it is a position-controller or effort-controller.
However, there are situations in which both of these low-level controls are not working well.
Are there any examples or guidelines for how to do this?

smoya23, 2024-04-11:
Hi @odj3460,

It's not impossible to implement low-level effort controller inside the current hardware_interface, it was just not part of our defined supported feature set. Our [example](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp) in the Kortex repo is a good starting point in order to get it to work inside our current hardware interface. Please note that you will also need to add the effort controller inside the ros2_controllers.yaml file of your arm description package (for a 6dof, it's located in [here]( https://github.com/Kinovarobotics/ros2_kortex/blob/main/kortex_description/arms/gen3/6dof/config/ros2_controllers.yaml)) . Regarding the error you're getting, is it possible for you to provide a code snippet of what is causing the error to happen? I can try and guide you from there.

Best,
Santiago

odj3460, 2024-04-22:
Thank you for the answer.

Actually, this problem was already solved.
But, I have one more question.

When I run the hardware_interface, I can get the topic for joint_states from ros2_controller. The topic is '/joint_states'.
The problem is the configuration of the joint values. 

From the topic, /joint_states is configured as [joint1 joint2 joint4 joint5 joint3 joint6]. Joint configuration order for joint 3-5 is wrong.
What's the reason for this problem?

aalmrad, 2024-10-08:
Hello, apologies for the late reply. We are recently trying to tackle the github issues on our repositories more actively. Apparently, the joints are listed in the right order everywhere in the relevant files and it seems that ros2_control is internally responsible for the order in which the joint states are being published and it is not clear how to control/modify this order. I do not know what part of your project requires the joint states in the correct order but did you try to reorder the joint states externally after receiving them from the /joint_states topic?

Best,
Abed