Kinovarobotics/ros2_kortex #203: Launching MoveIt Config fails

c-oechsner, 2024-01-23:
### Software Info
- Ubuntu 22.04.3 LTS (fresh install)
- ROS Rolling (fresh install according to https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html
- ros2_kortex (fresh install from binary according to https://github.com/Kinovarobotics/ros2_kortex/blob/main/README.md

### Robot Info
- Kinova Gen3 Ultra Lightweight Robot 6DoF
- Model Year 20
- Firmware Version 2.3.0-52

### What happens
I try to run `ros2 launch kinova_gen3_6dof_robotiq_2f_85_moveit_config robot.launch.py robot_ip:=10.163.18.200`. RViz starts but several errors occur (see below) so no planning is possible.

### What should happen
The planning scene should load and allow motion planning.

### Logs
`[move_group-8] [FATAL] [1706012699.612680496] [moveit.ros_planning.planning_pipeline]: Exception while loading planner 'chomp_interface/CHOMPPlanner': According to the loaded plugin descriptions the class chomp_interface/CHOMPPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  ompl_interface/OMPLPlanner pilz_industrial_motion_planner/CommandPlanner stomp_moveit/StompPlannerAvailable plugins: ompl_interface/OMPLPlanner, pilz_industrial_motion_planner/CommandPlanner, stomp_moveit/StompPlanner
[move_group-8] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[move_group-8]   what():  According to the loaded plugin descriptions the class chomp_interface/CHOMPPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  ompl_interface/OMPLPlanner pilz_industrial_motion_planner/CommandPlanner stomp_moveit/StompPlanner
[move_group-8] Stack trace (most recent call last):
[move_group-8] #16   Object "", at 0xffffffffffffffff, in 
[move_group-8] #15   Object "/opt/ros/rolling/lib/moveit_ros_move_group/move_group", at 0x558f460ebba4, in _start
[INFO] [spawner-5]: process has finished cleanly [pid 455890]
[move_group-8] #14   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7f0d5a429e3f]
[move_group-8] #13   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f0d5a429d8f]
[move_group-8] #12   Object "/opt/ros/rolling/lib/moveit_ros_move_group/move_group", at 0x558f460ea4a8, in main
[move_group-8] #11   Object "/opt/ros/rolling/lib/libmoveit_cpp.so.2.8.0", at 0x7f0d5b216fce, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&)
[move_group-8] #10   Object "/opt/ros/rolling/lib/libmoveit_cpp.so.2.8.0", at 0x7f0d5b216989, in moveit_cpp::MoveItCpp::loadPlanningPipelines(moveit_cpp::MoveItCpp::PlanningPipelineOptions const&)
[move_group-8] #9    Object "/opt/ros/rolling/lib/libmoveit_planning_pipeline_interfaces.so.2.8.0", at 0x7f0d5a7cd528, in moveit::planning_pipeline_interfaces::createPlanningPipelineMap(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-8] #8    Object "/opt/ros/rolling/lib/libmoveit_planning_pipeline.so.2.8.0", at 0x7f0d5abaee99, in planning_pipeline::PlanningPipeline::PlanningPipeline(std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-8] #7    Object "/opt/ros/rolling/lib/libmoveit_planning_pipeline.so.2.8.0", at 0x7f0d5aba3964, in 
[move_group-8] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f0d5a8ae52a, in __cxa_rethrow
[move_group-8] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f0d5a8ae276, in std::terminate()
[move_group-8] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f0d5a8ae20b, in 
[move_group-8] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f0d5a8a2b9d, in 
[move_group-8] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f0d5a4287f2]
[move_group-8] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f0d5a442475]
[move_group-8] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[move_group-8]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[move_group-8]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f0d5a4969fc]
[move_group-8] Aborted (Signal sent by tkill() 455896 948269429)
`

`[rviz2-10] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.`

`[rviz2-10] [ERROR] [1706013793.412730891] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_kinematics.manipulator.kinematics_solver_search_resolution' has invalid type: Wrong parameter type, parameter {robot_description_kinematics.manipulator.kinematics_solver_search_resolution} is of type {double}, setting it to {string} is not allowed.`

KarolineH, 2024-02-01:
Same error here. 
- using ROS Iron
- Kinova Gen3 Ultra Lightweight Robot 6DoF
- Model Year 23
- Firmware Version 2.5.2-1

It happens with the simulated robot as well, when you run `ros2 launch kinova_gen3_6dof_robotiq_2f_85_moveit_config robot.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true`

aalmrad, 2024-10-08:
Hello,

Apologies for the late reply. We are recently trying to address the github issues on our repositories more actively. We are currently focusing our resources on maintaining this repository for ROS2 Humble but we will adress the issues related to ROS2 rolling in the future. Thank you for your understanding.

Best,
Abed