Kinovarobotics/ros2_kortex #207: Error while compiling ros2 kortex with gazebo

alexandra1610, 2024-02-13:
Hi, I am trying to build from source and I keep getting the following error while compiling. Could you please help me solving this issue?
Thank you 

`colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`

> Starting >>> ros2_control_demo_description
> Starting >>> gazebo_ros2_control
> Starting >>> ros_gz_interfaces
> Starting >>> ros_gz_sim
> Starting >>> kortex_description                                              
> Starting >>> gz_ros2_control
> Starting >>> kortex_api
> Starting >>> ros2_control_demo_example_14
> Finished <<< ros2_control_demo_description [0.40s]                           
> Starting >>> ros2_control_demo_example_1
> Starting >>> ros2_control_demo_example_10
> Starting >>> ros2_control_demo_example_2
> Starting >>> ros2_control_demo_example_3                                  
> Starting >>> ros2_control_demo_example_4
> Starting >>> ros2_control_demo_example_5
> Starting >>> ros2_control_demo_example_6
> Starting >>> ros2_control_demo_example_7                                  
> Starting >>> ros2_control_demo_example_8
> Finished <<< kortex_description [0.65s]                                   
> Starting >>> kinova_gen3_6dof_robotiq_2f_85_moveit_config
> Starting >>> kinova_gen3_7dof_robotiq_2f_85_moveit_config                 
> Finished <<< kortex_api [0.81s]                                           
> Starting >>> kortex_driver
> Finished <<< ros_gz_sim [0.96s]                                           
> Finished <<< ros2_control_demo_example_14 [0.91s]
> Finished <<< ros2_control_demo_example_1 [1.19s]                          
> Finished <<< ros2_control_demo_example_2 [1.17s]                          
> Finished <<< ros2_control_demo_example_5 [1.11s]
> Finished <<< ros2_control_demo_example_10 [1.27s]                         
> Finished <<< ros2_control_demo_example_6 [1.14s]
> --- stderr: gazebo_ros2_control
> CMake Error at CMakeLists.txt:7 (find_package):
>   By not providing "Findgazebo_dev.cmake" in CMAKE_MODULE_PATH this project
>   has asked CMake to find a package configuration file provided by
>   "gazebo_dev", but CMake did not find one.
> 
>   Could not find a package configuration file provided by "gazebo_dev" with
>   any of the following names:
> 
>     gazebo_devConfig.cmake
>     gazebo_dev-config.cmake
> 
>   Add the installation prefix of "gazebo_dev" to CMAKE_PREFIX_PATH or set
>   "gazebo_dev_DIR" to a directory containing one of the above files.  If
>   "gazebo_dev" provides a separate development package or SDK, be sure it has
>   been installed.
> 
> 
> ---
> Failed   <<< gazebo_ros2_control [1.73s, exited with code 1]
> Aborted  <<< ros2_control_demo_example_8 [1.09s]
> Aborted  <<< ros2_control_demo_example_3 [1.25s]
> Aborted  <<< ros_gz_interfaces [1.70s]
> Aborted  <<< ros2_control_demo_example_4 [1.22s]
> Aborted  <<< ros2_control_demo_example_7 [1.12s]
> Aborted  <<< kinova_gen3_6dof_robotiq_2f_85_moveit_config [0.97s]
> Aborted  <<< kinova_gen3_7dof_robotiq_2f_85_moveit_config [0.94s]
> Aborted  <<< kortex_driver [0.79s]
> Aborted  <<< gz_ros2_control [4.35s]                                 
> 
> Summary: 10 packages finished [5.08s]
>   1 package failed: gazebo_ros2_control
>   9 packages aborted: gz_ros2_control kinova_gen3_6dof_robotiq_2f_85_moveit_config kinova_gen3_7dof_robotiq_2f_85_moveit_config kortex_driver ros2_control_demo_example_3 ros2_control_demo_example_4 ros2_control_demo_example_7 ros2_control_demo_example_8 ros_gz_interfaces
>   2 packages had stderr output: gazebo_ros2_control gz_ros2_control
>   11 packages not processed
>

moriarty, 2024-02-14:
The robotiq gripper uses mimic joints which are not merged into gazebo, likely this repos file is needed... branches may need to be updated depending on which distro of ROS you're using.

https://github.com/Kinovarobotics/ros2_kortex/blob/main/simulation.repos

alexandra1610, 2024-02-14:
Thank you, I downloaded this one by following the tutorial: 
vcs import src --skip-existing --input src/ros2_kortex/simulation.repos
I use ros2 humble 
But I still get the error. Should I delete the downloaded packages?

alexandra1610, 2024-02-14:
I have now a new error 
```
--- stderr: robotiq_driver
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findserial.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "serial", but
  CMake did not find one.

  Could not find a package configuration file provided by "serial" with any
  of the following names:

    serialConfig.cmake
    serial-config.cmake

  Add the installation prefix of "serial" to CMAKE_PREFIX_PATH or set
  "serial_DIR" to a directory containing one of the above files.  If "serial"
  provides a separate development package or SDK, be sure it has been
  installed.

```

alexandra1610, 2024-02-16:
I solved my issues following these steps that I wrote in a README file: 
https://git.initrobots.ca/amercader/kinova-kortex-installation

closed 2024-02-21 by alexandra1610