Kinovarobotics/ros2_kortex #208: Failed to activate controller when launching moveit

alexandra1610, 2024-02-21:
Hi, During the launch of ros2 launch kinova_gen3_7dof_robotiq_2f_85_moveit_config robot.launch.py   robot_ip:=yyy.yyy.yyy.yyy   use_fake_hardware:=true
I have an issue in the moveit plugin that there is no planning library loaded. I saw an error in the terminal:
Failed to activate controller
I attached the output of the terminal.
Thanks for helping
[error_launch_moveit.txt](https://github.com/Kinovarobotics/ros2_kortex/files/14365791/error_launch_moveit.txt)

timmarkhuff, 2024-05-08:
I encountered this same issue in simulation using humble. I ran 
```
ros2 launch kortex_bringup kortex_sim_control.launch.py \
  dof:=7 \
  use_sim_time:=true \
  launch_rviz:=false \
  vision:=true
```
and 
```
ros2 launch kinova_gen3_7dof_robotiq_2f_85_moveit_config sim.launch.py \
use_sim_time:=true
```
In RViz, I see a message indicating that it failed to load a planning library. I followed the instructions in this repo, so I'm not sure why it is not working.

alexandra1610, 2024-05-09:
Bonjour, oui il faut changer ompl_planning.yaml, utiliser celui qui est dans la librairie déjà compilée en l’installant avec sudo apt install et ensuite remplacer le fichier configuration. Cela m’a pris du temps mais c’est la solution que j’ai trouvé.Bon courage Alexandra Mercader

fzoric8, 2024-05-17:
@alexandra1610 Which config folder do you think?
moveit? 

Thanks!

aalmrad, 2024-10-08:
Closing due to inactivity - reply to the thread if you need the issue to be reopened

closed 2024-10-08 by aalmrad