Kinovarobotics/ros2_kortex #212: Licensing of the binary tarball downloaded by kortex_api

clalancette, 2024-03-19:
Hi there,
     I was recently looking into this package, and I noticed that https://github.com/Kinovarobotics/ros2_kortex/blob/cbfb28efb8c3007485aa80f86d692cd56e0e9109/kortex_api/CMakeLists.txt downloads https://artifactory.kinovaapps.com:443/artifactory/generic-public/kortex/API/2.5.0/linux_x86-64_x86_gcc.zip .  If I download and unpack that zip file, it has a bunch of headers and a couple of static libraries (`.a` files).  However, there is no mention in there of the license that all of that is released under.  Could you please give us some idea of what the license is?  Even better, could you add a license to that zip file so it would clear to future users under what terms the library comes?

Thank you.

smoya23, 2024-04-11:
Hi @clalancette,

You can find the lincence for our Kortex API inside our Github repo here : https://github.com/Kinovarobotics/kortex/blob/master/LICENSE

aalmrad, 2024-10-08:
Closing due to inactivity - reply to the thread if you need the issue to be reopened

closed 2024-10-08 by aalmrad