Kinovarobotics/ros2_kortex #213: Ignition simulation not loading controllers

Krawus, 2024-03-26:
Hi, when I'm launching simulation using ignition by command:
```
ros2 launch kortex_bringup kortex_sim_control.launch.py \
  dof:=7 \
  use_sim_time:=true \
  launch_rviz:=false
```
The robot "faints"
[image]

Logs from terminal sugesting that controllers are nod starting correctly:
```
[ruby $(which ign) gazebo-7] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-7] libEGL warning: egl: failed to create dri2 screen
[spawner-4] [INFO] [1711471385.182904319] [spawner_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711471385.184431476] [spawner_twist_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1711471385.185749050] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711471385.251607413] [spawner_robotiq_gripper_controller]: Waiting for '/controller_manager' node to exist
[ruby $(which ign) gazebo-7] [Err] [SystemLoader.cc:94] Failed to load system plugin [libign_ros2_control-system.so] : couldn't find shared library.
[ruby $(which ign) gazebo-7] [GUI] [Msg] Received world [empty.sdf] from the GUI.
[ruby $(which ign) gazebo-7] [Msg] Ignition Gazebo Server v6.16.0
[ruby $(which ign) gazebo-7] [Msg] Loading SDF world file[/usr/share/ignition/ignition-gazebo6/worlds/empty.sdf].
[ruby $(which ign) gazebo-7] [Msg] Serving entity system service on [/entity/system/add]
[ruby $(which ign) gazebo-7] [Msg] Create service on [/world/empty/create]
[ruby $(which ign) gazebo-7] [Msg] Remove service on [/world/empty/remove]
[ruby $(which ign) gazebo-7] [Msg] Pose service on [/world/empty/set_pose]
[ruby $(which ign) gazebo-7] [Msg] Pose service on [/world/empty/set_pose_vector]
[ruby $(which ign) gazebo-7] [Msg] Light configuration service on [/world/empty/light_config]
[ruby $(which ign) gazebo-7] [Msg] Physics service on [/world/empty/set_physics]
[ruby $(which ign) gazebo-7] [Msg] SphericalCoordinates service on [/world/empty/set_spherical_coordinates]
[ruby $(which ign) gazebo-7] [Msg] Enable collision service on [/world/empty/enable_collision]
[ruby $(which ign) gazebo-7] [Msg] Disable collision service on [/world/empty/disable_collision]
[ruby $(which ign) gazebo-7] [Msg] Material service on [/world/empty/visual_config]
[ruby $(which ign) gazebo-7] [Msg] Material service on [/world/empty/wheel_slip]
[ruby $(which ign) gazebo-7] [Msg] Loaded level [3]
[ruby $(which ign) gazebo-7] [Msg] Serving world controls on [/world/empty/control], [/world/empty/control/state] and [/world/empty/playback/control]
[ruby $(which ign) gazebo-7] [Msg] Serving GUI information on [/world/empty/gui/info]
[ruby $(which ign) gazebo-7] [Msg] World [empty] initialized with [1ms] physics profile.
[ruby $(which ign) gazebo-7] [Msg] Serving world SDF generation service on [/world/empty/generate_world_sdf]
[ruby $(which ign) gazebo-7] [Msg] Serving world names on [/gazebo/worlds]
[ruby $(which ign) gazebo-7] [Msg] Resource path add service on [/gazebo/resource_paths/add].
[ruby $(which ign) gazebo-7] [Msg] Resource path get service on [/gazebo/resource_paths/get].
[ruby $(which ign) gazebo-7] [Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[ruby $(which ign) gazebo-7] [Msg] Resource paths published on [/gazebo/resource_paths].
[ruby $(which ign) gazebo-7] [Msg] Server control service on [/server_control].
[ruby $(which ign) gazebo-7] [Msg] Found no publishers on /stats, adding root stats topic
[ruby $(which ign) gazebo-7] [Msg] Found no publishers on /clock, adding root clock topic
[ruby $(which ign) gazebo-7] [Msg] Serving scene information on [/world/empty/scene/info]
[ruby $(which ign) gazebo-7] [Msg] Serving graph information on [/world/empty/scene/graph]
[ruby $(which ign) gazebo-7] [Msg] Serving full state on [/world/empty/state]
[ruby $(which ign) gazebo-7] [Msg] Serving full state (async) on [/world/empty/state_async]
[ruby $(which ign) gazebo-7] [Msg] Publishing scene information on [/world/empty/scene/info]
[ruby $(which ign) gazebo-7] [Msg] Publishing entity deletions on [/world/empty/scene/deletion]
[ruby $(which ign) gazebo-7] [Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v125WorldE], which doesn't have `operator<<`. Component will not be serialized.
[ruby $(which ign) gazebo-7] [GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v125WorldE], which doesn't have `operator>>`. Component will not be deserialized.
[spawner-3] [INFO] [1711471387.207151437] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711471387.207160561] [spawner_twist_controller]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1711471387.207163743] [spawner_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711471387.269412988] [spawner_robotiq_gripper_controller]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1711471389.236372709] [spawner_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1711471389.236654569] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711471389.239586830] [spawner_twist_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711471389.297994496] [spawner_robotiq_gripper_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1711471391.261493054] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1711471391.262223452] [spawner_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711471391.267709179] [spawner_twist_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711471391.323219582] [spawner_robotiq_gripper_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [ERROR] [1711471393.290904503] [spawner_joint_state_broadcaster]: Controller manager not available
[spawner-4] [ERROR] [1711471393.291395294] [spawner_joint_trajectory_controller]: Controller manager not available
[spawner-5] [ERROR] [1711471393.295700689] [spawner_twist_controller]: Controller manager not available
[spawner-6] [ERROR] [1711471393.352222531] [spawner_robotiq_gripper_controller]: Controller manager not available

```
Is this somehow a known issue? Thank you in advance for your help

smoya23, 2024-04-04:
Hello @Krawus,

This is not an expected situation. I see from your logs that there seems to be a missing lib on your system

`[ruby $(which ign) gazebo-7] [Err] [SystemLoader.cc:94] Failed to load system plugin [libign_ros2_control-system.so] : couldn't find shared library.`

Did you install the package from binaries or you built it from source? Either way, make sure that you have correctly installed Ingition. This may guide may be helpful : https://gazebosim.org/api/gazebo/6.1/install.html. Once you have this installed, try launching the simulation again, you should then see the arm in candle position.

[image]

Best,
Santiago

diyaagarwal21, 2024-09-02:
Hi,
I am also having the same error: 

```
[ros2-9] [INFO] [1725293330.569761463] [_ros2cli_5295]: waiting for service /controller_manager/load_controller to become available...
[ign-6] [Err] [SystemLoader.cc:94] Failed to load system plugin [libign_ros2_control-system.so] : couldn't find shared library.
[ros2-10] [WARN] [1725293339.878516819] [_ros2cli_5297]: Could not contact service /controller_manager/load_controller
```

The link https://gazebosim.org/api/gazebo/6.1/install.html does not work (says page not found). And I also did install the plugin libign_ros2_control-system.so and its package.

I am using ROS2 Humble and Gazebo Ignition (fortress). My quadruped robot (Spot robot) spawns on Gazebo, but I cannot get it to walk because of the controller manager issue. I am also not using a ros_gz_bridge. How can I fix this issue? Thank you!

aalmrad, 2024-09-03:
Hello, 

The missing shared library should be found under /opt/ros/humble/lib on your system. If this is not the case, please make sure to install ign_ros2_control package using the following command: `sudo apt-get install ros-humble-ign-ros2-control`

I hope this solves the issue and please keep me posted regarding the situation.

Best,
Abed

diyaagarwal21, 2024-09-04:
Hi, 

Thank you! I was able to fix the issue by launching Ignition Gazebo with this command: 
```
PythonLaunchDescriptionSource(
                [os.path.join(get_package_share_directory('ros_ign_gazebo'),
                              'launch', 'ign_gazebo.launch.py')]),
            launch_arguments=[('gz_args', [' -r -v 4 ', ground_plane_path])]),
```
where `ground_plane_path` is the path to the sdf world file. The plugins are working and controllers are loading. However, I am still not able to get the robot to walk. I tried using a Twist command that published well but the robot didn't move (though everything seemed connected well in the rqt graph), and trying to control the joint trajectory directly didn't work because of a clock issue that I am unable to fix. Do you by any chance have any recommendations on how to get the robot to walk in the simulation? Thank you so much.

aalmrad, 2024-10-04:
Hello @diyaagarwal21 ,

How are you exactly integrating the Gen3 with Spot? I have no details about your project and this does not allow me to provide a well tailored solution. But as general guidelines, if you are facing time synchronization issues between ROS and the Gazebo, make sure the set the use_sim_time parameter to true. 
Furthermore, I recommend that you make sure that all of your controllers are active using: 
```
ros2 control list_controllers --controller-manager
```
Moreover, double check if the command interfaces of the controllers accept velocity and position commands.

Best,
Abed

meddy-jaka, 2025-04-14:
Hi @diyaagarwal21 

I encountered the same clock issue before. The key fix was to declare and use the simulation clock by enabling the `use_sim_time` parameter. This needs to be passed as a launch configuration parameter to your **move_group** launch as well as to other critical nodes such as the **Gazebo spawning node** and the **controller manager**.

For example, you can declare the `use_sim_time` parameter and include it in your move_group parameters as follows:

```
ld.add_action(
    DeclareBooleanLaunchArg(
        "use_sim_time", default_value=True, description="Use simulation time"
    )
)

move_group_params = [
    moveit_config.to_dict(),
    move_group_configuration,
    {"use_sim_time": LaunchConfiguration("use_sim_time")},
]
```

For more detailed information and a complete working example, you can check out our full implementation on GitHub:
https://github.com/JakaCobot/jaka_ros2