Kinovarobotics/ros2_kortex #216: Get the error Invalid parameter "isaac_joint_commands" when building from source

Reagan1311, 2024-04-24:
Hi there, the following error is encountered when executing the command in the Usage section. I followed the instructions to build from source step by step. Any idea why?

[ERROR] [launch]: Caught exception in launch (see debug for traceback): Invalid parameter "isaac_joint_commands"

smoya23, 2024-04-29:
Hi @Reagan1311,

Can you try to build again but following the updated instructions in this [PR](https://github.com/Kinovarobotics/ros2_kortex/pull/215) ? There are some missing steps in the current instructions, which I'm pretty sure is the cause of your error. Let me know how it goes and if you get any other error.

Best,
Santiago

smoya23, 2024-05-03:
PR was merged, you should be able to make it work now by following the instructions of our ReadMe file

Reagan1311, 2024-05-03:
Thanks a lot, will try it again

closed 2024-05-03 by Reagan1311