Kinovarobotics/ros2_kortex #217: `gazebo_ros2_control` not available on Ubuntu 24.04 Jazzy / Rolling

sea-bass, 2024-05-03:
I just updated to Ubuntu 24.04 / ROS 2 Rolling (but Jazzy would do as well), and running into issues with the dependency in the `kortex_bringup` package:

```xml
<exec_depend>gazebo_ros2_control</exec_depend>
```

Since Gazebo classic is not available going forward, I think this dependency needs to be removed to the `main` branch, as well as things that use that dependency.

smoya23, 2024-05-03:
Hi @sea-bass ,

I'm not sure what do you mean that Gazebo classic is not available moving forward, it looks like it is being ported into Rolling according to [this](https://github.com/ros/rosdistro/blob/14b7539a0112dfb01a2496f4d54c76a8a45fe63d/rolling/distribution.yaml#L1660) package list mentionned by the ROS Discourse's [thread](https://discourse.ros.org/t/preparing-ros-2-rolling-for-the-transition-to-ubuntu-24-04/35673/17) about migration to Ubuntu 24.04 / Rolling. We are waiting for the simulation packages to be released and once it has been done, we will for sure evaluate what needs to be done to use our package on Rolling

sea-bass, 2024-05-03:
Oh, then I guess it's just not yet available from binaries. 

But then building this package from source also requires the `gazebo-dev` libraries which also were not yet available in the testing repos. Depending on whether those will also be added soon, things may be okay.

Thanks!

smoya23, 2024-05-03:
No problem, I will close this issue for now. We will open a PR in due time once the migration of simulation packages is done.

Cheers!

closed 2024-05-03 by smoya23

sea-bass, 2024-07-04:
So... it seems they decided this package this package is not going to ever be released for Jazzy onwards, meaning `rosdep install` will fail on those distros due to this dependency. See here:

https://control.ros.org/master/doc/gazebo_ros2_control/doc/index.html

I suggest reopening this issue and removing the Gazebo classic support on main. I will contribute this PR if the maintainers are OK with this plan.