Kinovarobotics/ros2_kortex #219: Invalid parameter: "com_port"

htchr, 2024-05-17:
Hello,

I am trying to get my Kinova Gen 3 Lite 6 dof arms working on ROS 2 Humble.
I have created a new devcontainer environment in VSCode for ROS 2 Humble and installed everything as instructed on the main branch.
When I try to launch the example in the README:

```bash
ros2 launch kortex_bringup gen3.launch.py \
  robot_ip:=yyy.yyy.yyy.yyy \
  use_fake_hardware:=true \
  robot_type:=gen3_lite \
  gripper:=gen3_lite_2f \
  dof:=6
```

I get the following output: 

```bash
vscode@ubu22:/colcon_ws$ ros2 launch kortex_bringup gen3.launch.py \
  robot_ip:=yyy.yyy.yyy.yyy \
  use_fake_hardware:=true \
  robot_type:=gen3_lite \
  gripper:=gen3_lite_2f \
  dof:=6
[INFO] [launch]: All log files can be found below /home/vscode/.ros/log/2024-05-17-14-41-15-504013-ubu22-51737
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command failed. Command: /opt/ros/humble/bin/xacro /colcon_ws/install/kortex_description/share/kortex_description/robots/gen3.xacro robot_ip:=yyy.yyy.yyy.yyy name:=arm arm:=gen3_lite dof:=6 prefix:="" use_fake_hardware:=true fake_sensor_commands:=false gripper:=gen3_lite_2f use_internal_bus_gripper_comm:=false gripper_max_velocity:=100.0 gripper_max_force:=100.0 gripper_joint_name:=robotiq_85_left_knuckle_joint 
Captured stderr output: error: Invalid parameter "com_port"
when instantiating macro: load_gripper (/colcon_ws/install/kortex_description/share/kortex_description/grippers/gen3_lite_2f/urdf/gen3_lite_2f_macro.xacro)
instantiated from: load_robot (/colcon_ws/install/kortex_description/share/kortex_description/robots/kortex_robot.xacro)
in file: /colcon_ws/install/kortex_description/share/kortex_description/robots/gen3.xacro

vscode@ubu22:/colcon_ws$ ros2 launch kortex_bringup gen3.launch.py \        robot_ip:=10.18.2.239 \
  use_fake_hardware:=false \
  robot_type:=gen3_lite \
  gripper:=gen3_lite_2f \
  dof:=6
[INFO] [launch]: All log files can be found below /home/vscode/.ros/log/2024-05-17-14-41-43-276849-ubu22-51918
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command failed. Command: /opt/ros/humble/bin/xacro /colcon_ws/install/kortex_description/share/kortex_description/robots/gen3.xacro robot_ip:=10.18.2.239 name:=arm arm:=gen3_lite dof:=6 prefix:="" use_fake_hardware:=false fake_sensor_commands:=false gripper:=gen3_lite_2f use_internal_bus_gripper_comm:=false gripper_max_velocity:=100.0 gripper_max_force:=100.0 gripper_joint_name:=robotiq_85_left_knuckle_joint 
Captured stderr output: error: Invalid parameter "com_port"
when instantiating macro: load_gripper (/colcon_ws/install/kortex_description/share/kortex_description/grippers/gen3_lite_2f/urdf/gen3_lite_2f_macro.xacro)
instantiated from: load_robot (/colcon_ws/install/kortex_description/share/kortex_description/robots/kortex_robot.xacro)
in file: /colcon_ws/install/kortex_description/share/kortex_description/robots/gen3.xacro
```

Has anyone been able to successfully run a Kinova Gen 3 Lite 6 dof arm in ROS 2 Humble?

Thank you,

smoya23, 2024-05-21:
Hi @htchr,

I recently created a branch to fix this issue, can you check out branch `pr_gen3-lite-launch-file` and try again ? Please make sure to read again the README file as I also updated it.

htchr, 2024-05-22:
Hello @smoya23, the branch seems to work well from preliminary testing.
Thank you!

closed 2024-05-22 by htchr