Kinovarobotics/ros2_kortex #221: Gen3_lite has problem in gazebo simulation.

htcb0249, 2024-05-24:
I tried the branch pr_gen3-lite-launch-file, but it shows sim_isaac invalid parameter message. The gen3 model also has a problem in Rviz window This sim_isaac parameter is in the description document. 
```shell
ros2 launch kortex_bringup kortex_sim_control.launch.py   dof:=6   robot_type:=gen3_lite   use_sim_time:=true  
[INFO] [launch]: All log files can be found below /home/ros2/.ros/log/2024-05-24-08-52-31-362799-ros2-virtual-kinova-2884
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command failed. Command: /opt/ros/humble/bin/xacro /opt/ros/humble/share/kortex_description/robots/kinova.urdf.xacro robot_ip:=xxx.yyy.zzz.www name:=gen3 arm:=gen3_lite dof:=6 vision:=false prefix:="" sim_gazebo:=false sim_ignition:=true simulation_controllers:=/opt/ros/humble/share/kortex_description/arms/gen3/6dof/config/ros2_controllers.yaml gripper:=robotiq_2f_85 
Captured stderr output: error: Invalid parameter "sim_isaac"
when instantiating macro: load_arm (/opt/ros/humble/share/kortex_description/arms/gen3_lite/6dof/urdf/gen3_lite_macro.xacro)
instantiated from: load_robot (/opt/ros/humble/share/kortex_description/robots/kortex_robot.xacro)
in file: /opt/ros/humble/share/kortex_description/robots/kinova.urdf.xacro
```
While using command
```shell
ros2 launch kortex_bringup kortex_sim_control.launch.py \
  dof:=6 \
  robot_type:=gen3_lite \
  use_sim_time:=true \
  launch_rviz:=false \
  sim_gazebo:=true 
```
[image]

smoya23, 2024-05-24:
Hi @htcb0249 ,
This is because there is some dependencies missing in your workspace, could you redo the commands inside our building from source [instructions](https://github.com/Kinovarobotics/ros2_kortex?tab=readme-ov-file#contributing-to-this-repository-or-building-from-source) (it was recently updated). Once you have built your workspace, don't forget to source it before running the command again.

```
cd <your-ros2-rowkspace>
source install/setup.bash
```
Cheers,
Santiago

htcb0249, 2024-05-27:
No, you are misunderstanding what I mean. I've sourced my setup.bash (I'm omitting cd and source commands), and it shows these messages.

Does that updated "main" branch conflict with some packages in [ros2 tutorials](https://moveit.picknik.ai/main/doc/tutorials/tutorials.html)? I can't build the workspace. When I tried to build them with colcon, it always shows messages like: 
```bash
--- stderr: moveit_task_constructor_core    
/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp: In constructor ‘moveit::task_constructor::solvers::PipelinePlanner::PipelinePlanner(const SharedPtr&, const string&)’:
/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:128:104: error: ‘DISPLAY_PATH_TOPIC’ is not a member of ‘planning_pipeline::PlanningPipeline’
  128 | solutions on topic " + planning_pipeline::PlanningPipeline::DISPLAY_PATH_TOPIC);
      |                                                             ^~~~~~~~~~~~~~~~~~

/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:131:66: error: ‘MOTION_PLAN_REQUEST_TOPIC’ is not a member of ‘planning_pipeline::PlanningPipeline’
  131 |                        planning_pipeline::PlanningPipeline::MOTION_PLAN_REQUEST_TOPIC);
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~

/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp: In member function ‘virtual void moveit::task_constructor::solvers::PipelinePlanner::init(const RobotModelConstPtr&)’:
/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:151:45: warning: ‘void planning_pipeline::PlanningPipeline::displayComputedMotionPlans(bool)’ is deprecated: Use generatePlan or ROS parameter API instead. [-Wdeprecated-declarations]
  151 |         planner_->displayComputedMotionPlans(properties().get<bool>("display_motion_plans"));
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:42:
/home/ros2/ws_moveit/install/moveit_ros_planning/include/moveit_ros_planning/moveit/planning_pipeline/planning_pipeline.h:134:73: note: declared here
  134 | ed("Use generatePlan or ROS parameter API instead.")]] void displayComputedMotionPlans(bool /*flag*/){};
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~

/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:152:42: warning: ‘void planning_pipeline::PlanningPipeline::publishReceivedRequests(bool)’ is deprecated: Use generatePlan or ROS parameter API instead. [-Wdeprecated-declarations]
  152 |         planner_->publishReceivedRequests(properties().get<bool>("publish_planning_requests"));
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:42:
/home/ros2/ws_moveit/install/moveit_ros_planning/include/moveit_ros_planning/moveit/planning_pipeline/planning_pipeline.h:135:73: note: declared here
  135 | ed("Use generatePlan or ROS parameter API instead.")]] void publishReceivedRequests(bool /*flag*/){};
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~

/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp: In member function ‘moveit::task_constructor::solvers::PlannerInterface::Result moveit::task_constructor::solvers::PipelinePlanner::plan(const PlanningSceneConstPtr&, const MotionPlanRequest&, robot_trajectory::RobotTrajectoryPtr&)’:
/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:213:22: error: ‘struct planning_interface::MotionPlanResponse’ has no member named ‘trajectory_’; did you mean ‘trajectory’?
  213 |         result = res.trajectory_;
      |                      ^~~~~~~~~~~
      |                      trajectory
/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:214:67: error: ‘error_code_to_string’ is not a member of ‘moveit::core’; did you mean ‘errorCodeToString’?
  214 |   return { success, success ? std::string() : moveit::core::error_code_to_string(res.error_code_.val) };
      |                                                             ^~~~~~~~~~~~~~~~~~~~
      |                                                             errorCodeToString
/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:214:92: error: ‘struct planning_interface::MotionPlanResponse’ has no member named ‘error_code_’; did you mean ‘error_code’?
  214 | ss ? std::string() : moveit::core::error_code_to_string(res.error_code_.val) };
      |                                                             ^~~~~~~~~~~
      |                                                             error_code
/home/ros2/ws_moveit/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:214:109: error: could not convert ‘{success, <expression error>}’ from ‘<brace-enclosed initializer list>’ to ‘moveit::task_constructor::solvers::PlannerInterface::Result’
  214 | ::string() : moveit::core::error_code_to_string(res.error_code_.val) };
      |                                                                      ^
[... cut ...]
  ros2 topic pub /joint_trajectory_controller/joint_trajectory trajectory_msgs/msg/JointTrajectory "{
    header: {stamp: now},
    joint_names: ['joint_1', 'joint_2', 'joint_3', 'joint_4', 'joint_5', 'joint_6'],
    points: [
      {
        positions: [0.0, 0.5, 1.0, 0.5, 0.5, 0.2],
        time_from_start: {sec: 5}
      }
    ]
  }" -1
```
Terminal shows:
```bash
[ruby $(which ign) gazebo-7] [INFO] [1716793015.871516302] [controller_manager]: Empty activate and deactivate list, not requesting switch
```

smoya23, 2024-05-27:
Hi @htcb0249, 

Moveit task constructor is not part of our dependencies, so something is wrong in your workspace. These are the dependencies that should be present in your workspace. 

[image]

Note that I have installed moveit2 from source directly inside my workspace. At the end of the build you should see 114 packages built, not 59

[image]

Also, I don't know which ROS2 distro you're on, but just in case, note that currently our driver is not compatible with ROS 2 Rolling, as Rolling has been ported to Ubuntu 24.04 and all of the simulation packages have not been ported yet. I see from your build commands that you're also missing the command to import the simulation packages 

```
vcs import src --skip-existing --input src/ros2_kortex/simulation.humble.repos
```
Which would explain why the Gen3_Lite is appearing on Gazebo but collapsing with gravity (gazebo controllers are part of the simulation packages that I suspect are missing)

htcb0249, 2024-05-28:
In my **Ubuntu 22.04 ROS2 humble distro** workspace,  the dependencies are as follows:
[image]
Then I execute the commands you provided.
[image]
There are only **63** packages, that might be the problem.
[image]

smoya23, 2024-05-28:
@htcb0249  63 is the correct number of packages if you don't have installed moveit, i had 114 packages as I installed moveit from [source](https://moveit.ros.org/install-moveit2/source/) directly in my workspace. If you don't plan to use Moveit, then your workspace looks fine. Does your build finish without issues ? If you re-run the simulation launch file, are you able to see the Gen3_lite in Gazebo/Ignition ?

Also, please note that our driver does not include the Gen3_lite moveit config, so if you wish to use a Gen3_lite arm with MoveIt, you will need to generate it yourself using the Moveit assistant GUI.

htcb0249, 2024-05-29:
Thank you, I can run simulation to see Gen3_lite in Gazebo/Ignition now. But I don't know how to control it to move. 
[image]

I attempted to build my workspace using ROS2 Kortex and MoveIt 2 to control my simulated arm before transitioning to my real robotic arm (Gen3 Lite), which is currently under maintenance.

Although I noticed that the MoveIt 2 tutorials in the Humble branch use different examples such as the Quick Start, which might be causing conflicts, I will attempt to replace some files from the main branch to see if that resolves the issue.

Based on your response, I understand that I cannot use MoveIt 2 to design the motion for my Gen3_Lite. Is it possible to use the Gen3 as a substitute?

htcb0249, 2024-05-31:
Ok, I got it. That means all I need to do is to set up using Moveit's Setup Assistant, and it is the normal way to use the Gen3 Lite with MoveIt. I'll try it. Thank you.

closed 2024-05-31 by htcb0249