Kinovarobotics/ros2_kortex #222: ros2_control_node segfault when running robot_bringup.launch on real robot

fzoric8, 2024-05-28:
Hi, 

I'm experiencing segfault when running `robot_bringup.launch` from `ros2_kortex` for the ros2 `humble`: 

Error output is: 
```
Creating session for communication
[ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-1]   what():  not connected !!!
[ros2_control_node-1] Stack trace (most recent call last):
[ros2_control_node-1] #19   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-1] #18   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x5648385e7d84, in 
[ros2_control_node-1] #17   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8ac015ae3f, in __libc_start_main
[ros2_control_node-1] #16   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8ac015ad8f, in 
[ros2_control_node-1] #15   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x5648385e789e, in 
[ros2_control_node-1] #14   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x7f8ac084fe79, in controller_manager::ControllerManager::ControllerManager(std::shared_ptr<rclcpp::Executor>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&)
[ros2_control_node-1] #13   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x7f8ac084b6bf, in controller_manager::ControllerManager::init_resource_manager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ros2_control_node-1] #12   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f8ac005dd9d, in hardware_interface::ResourceManager::load_urdf(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool)
[ros2_control_node-1] #11   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f8ac005ce7b, in 
[ros2_control_node-1] #10   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f8ac0084caa, in 
[ros2_control_node-1] #9    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f8ac00831db, in hardware_interface::System::initialize(hardware_interface::HardwareInfo const&)
[ros2_control_node-1] #8    Object "/root/ws_moveit2/build/kortex_driver/libkortex_driver.so", at 0x7f8aadc1f00f, in kortex_driver::KortexMultiInterfaceHardware::on_init(hardware_interface::HardwareInfo const&)
[ros2_control_node-1] #7    Object "/root/ws_moveit2/build/kortex_driver/libkortex_driver.so", at 0x7f8aadcf5f02, in Kinova::Api::SessionManager::CreateSession(Kinova::Api::Session::CreateSessionInfo const&)
[ros2_control_node-1] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8ac04284d7, in __cxa_throw
[ros2_control_node-1] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8ac0428276, in std::terminate()
[ros2_control_node-1] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8ac042820b, in 
[ros2_control_node-1] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8ac041cb9d, in 
[ros2_control_node-1] #2    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8ac01597f2, in abort
[ros2_control_node-1] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8ac0173475, in raise
[ros2_control_node-1] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8ac01c79fc, in pthread_kill
[ros2_control_node-1] Aborted (Signal sent by tkill() 13675 0)
```

I'm aware that this may be `ros2_control_node` problem, but if anyone had same problem, any help would be greatily appreciated.

smoya23, 2024-05-28:
Hi @fzoric8 ,

Can you provide more details on what you're trying to do? Which command did you run specifically, the `robot_bringup.launch` file does not exist. Do you have a real robot or are you trying to use a gazebo simulated/fake hardware one?

fzoric8, 2024-05-28:
Hi @smoya23, 

I'm tryng to launch `gen3` robot on real robot. 
I've figured out that segmentation fault occurs because my robot was connected over ethernet to my laptop dock. When I connected it directly to my laptop, there's no segmentation fault. 

However, following warning appear: 
```
[spawner-5] [INFO] [1716907416.817350453] [spawner_twist_controller]: Waiting for '/controller_manager' services to be available
[spawner-3] [INFO] [1716907416.825386875] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-6] [INFO] [1716907416.826389381] [spawner_robotiq_gripper_controller]: Waiting for '/controller_manager' services to be available
[spawner-4] [INFO] [1716907418.813499985] [spawner_joint_trajectory_controller]: Waiting for '/controller_manager' services to be available
[spawner-5] [INFO] [1716907418.845475542] [spawner_twist_controller]: Waiting for '/controller_manager' services to be available
```

when running: 
```
ros2 launch kortex_bringup gen3.launch.py robot_ip:=192.168.1.10 
```

closed 2024-06-18 by fzoric8