Kinovarobotics/ros2_kortex #224: Issue with gen3.launch.py loading default end effector despite blank gripper argument

keenan88, 2024-06-13:
Hello,

I am working with the gen3 robot and trying to run gen3.launch.py without any end effector. When I launch gen3.launch.py from my launch file with the gripper argument set to a blank string, the robot is loaded with an end effector [kortex_interface_output.txt](https://github.com/user-attachments/files/15828678/kortex_interface_output.txt). I am using a custom end effector, and therefore would not like to load any gripper. I installed ros2_kortex with the binaries, using:
apt install ros-humble-kortex-bringup -y
apt install ros-humble-kinova-gen3-7dof-robotiq-2f-85-moveit-config -y

Launch file: [kortex_interface_launch.txt](https://github.com/user-attachments/files/15828697/kortex_interface_launch.txt)
Project repository: https://github.com/keenan88/isaacsim_ros2_greenhouse

Thank you,

Keenan

aalmrad, 2024-06-14:
Hello Keenan,
your question was well received and we are currently working on finding a solution :)

aalmrad, 2024-06-17:
After investigating the issue, it seems that our repository is not designed to work with a robotic arm without any gripper. However, we will push a new update to the repository very soon to address this request. In the meantime, we provide the option to build the ros2 packages for Gen3 from source, so feel free to apply the modifications yourself in the meantime.
Here is the modified "gen3.launch.py" file:

```python
# Copyright (c) 2021 PickNik, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# Author: Denis Stogl

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir

def generate_launch_description():
    # Declare arguments
    declared_arguments = []
    declared_arguments.append(
        DeclareLaunchArgument(
            "robot_type",
            default_value="gen3",
            description="Type/series of robot.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "robot_ip",
            description="IP address by which the robot can be reached.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument("dof", default_value="7", description="DoF of robot.")
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "use_fake_hardware",
            default_value="false",
            description="Start robot with fake hardware mirroring command to its states.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "fake_sensor_commands",
            default_value="false",
            description="Enable fake command interfaces for sensors used for simple simulations. \
            Used only if 'use_fake_hardware' parameter is true.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "robot_controller",
            default_value="joint_trajectory_controller",
            description="Robot controller to start.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "controllers_file",
            default_value="ros2_controllers.yaml",
[... 61 lines cut ...]
            "robot_type": robot_type,
            "robot_ip": robot_ip,
            "dof": dof,
            "use_fake_hardware": use_fake_hardware,
            "fake_sensor_commands": fake_sensor_commands,
            "robot_controller": robot_controller,
            "gripper": gripper,
            "use_internal_bus_gripper_comm": use_internal_bus_gripper_comm,
            "gripper_max_velocity": gripper_max_velocity,
            "gripper_max_force": gripper_max_force,
            "gripper_joint_name": gripper_joint_name,
            "launch_rviz": launch_rviz,
            "controllers_file": controllers_file,
            "description_file": "gen3.xacro",
        }.items(),
    )

    return LaunchDescription(declared_arguments + [base_launch])

``` 
As you can see the **choices=["robotiq_2f_85", "robotiq_2f_140"],** line for the gripper argument was commented out and as such, you can now pass an empty value for the gripper argument from the terminal either as "<Space Character>" or '""' so the terminal command line would look like the following if you're not connected to a real-life Gen3: 
```shell
ros2 launch kortex_bringup gen3.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true gripper:='""'
``` 
Please make sure to remove your previous binary installation first and to build the corresponding packages after modifying any launch file. I hope this will help you out :)

keenan88, 2024-06-18:
Hello, 

Thank you for your response. I have installed and built the source code following the steps listed in the latest commit in main (33ec22cd), and replaced gen3.launch.py. 

`ros2 launch kortex_bringup gen3.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true gripper:='""'`

launches without error, but the robot is stilled launched with the robotiq_85 gripper. 

([v2_gen3_launch_output.txt](https://github.com/user-attachments/files/15887901/v2_gen3_launch_output.txt))

[image]

Are there further edits required to the source code?

keenan88, 2024-06-18:
I have made the following modifications:

**gen3.xacro:**
Changed the gripper in load_robot macro to be always set to "" instead of $(arg gripper).

**kortex_control.launch.py:**
Commented out the robot_hand_controller_spawner from nodes_to_launch list.

As a result, the arm is now visible in RVIZ without the end effector when running:

`ros2 launch kortex_bringup gen3.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true gripper:='""'`

[image]

[v3_gen3_launch_output.txt](https://github.com/user-attachments/files/15888462/v3_gen3_launch_output.txt)

I have not yet validated movement on the physical arm.

aalmrad, 2024-10-07:
Hello @keenan88,

Apologies for the late reply, we are recently trying to be more active in tackling the issues on our github repositories. This issue was resolved in the following PR: [242](https://github.com/Kinovarobotics/ros2_kortex/pull/242). In this PR, the gripper default value was set to an empty string and the gripper controller was set to be activated only if a gripper is specified. I will close this issue and please do not hesitate to reopen the thread if you face further problems.

Best,
Abed