Kinovarobotics/ros2_kortex #225: Build error on aarch64 architecture (Jetson AGX Orin)

mehradmrt, 2024-06-18:
The kortex_api package is not built properly with colcon. Replaced the API_URL in the package CMakeLists.txt with 'https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_aarch64_gcc_7.4.zip' but still not able to build the package properly. What is a potential solution?

Would it be possible to run the robot without building kortex_api?

martinleroux, 2024-06-18:
Hi @mehradmrt ,

It is not possible to run the robot via ROS2 without building kortex_api. You can of course drive the robot from the Web App GUI, or use kortex_api in Python outside of ROS to avoid having to build altogether.

Can you elaborate on the exact problem you are seeing? We'll see if we can assist you with the process.

mehradmrt, 2024-06-18:
Hi @martinleroux, thanks for your reply. I need it to be run on MoveIt2 and ROS2 humble as all our sensors and drivers are running on ROS. We don't need the API at this point so if there would be any workaround to run the robot on ROS2/MoveIt that works as well.

mehradmrt, 2024-06-18:
I received an error building the kortex_api. Please note that I have replaced the API_URL in the package CMakeLists.txt with 'https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0/linux_aarch64_gcc_7.4.zip:   

```colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
[1.436s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'ros2_control_test_assets' is in: /opt/ros/humble
	'control_msgs' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding control_msgs ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> control_msgs
Starting >>> ros2_control_test_assets
Starting >>> controller_manager_msgs
Starting >>> ros_gz_interfaces
Starting >>> joint_limits
Starting >>> ros_gz_sim
Starting >>> kortex_api
Starting >>> serial
Starting >>> ros2_controllers_test_nodes
Finished <<< ros2_control_test_assets [0.90s]                     
Finished <<< ros2_controllers_test_nodes [2.16s]                  
Finished <<< serial [4.13s]                                       
--- stderr: kortex_api                                                 
[ 11%] Creating directories for 'kinova_binary_api-populate'
[ 22%] Performing download step (download, verify and extract) for 'kinova_binary_api-populate'
-- Downloading...
   dst='/robot_ws/kortex_ros2_ws/build/kortex_api/_deps/kinova_binary_api-subbuild/kinova_binary_api-populate-prefix/src/linux_aarch64_gcc_7.4.zip'
   timeout='none'
   inactivity timeout='none'
-- Using src='https://artifactory.kinovaapps.com:443/artifactory/generic-public/kortex/API/2.6.0/linux_aarch64_gcc_7.4.zip'
-- [download 0% complete]
-- [download 1% complete]
...
-- [download 100% complete]
-- verifying file...
       file='../robot_ws/kortex_ros2_ws/build/kortex_api/_deps/kinova_binary_api-subbuild/kinova_binary_api-populate-prefix/src/linux_aarch64_gcc_7.4.zip'
-- MD5 hash of
    /robot_ws/kortex_ros2_ws/build/kortex_api/_deps/kinova_binary_api-subbuild/kinova_binary_api-populate-prefix/src/linux_aarch64_gcc_7.4.zip
  does not match expected value
    expected: '64bd86e7ab8bda90ef1fc7d6a356e080'
      actual: 'd7771a7c106888717ace359164d40235'
-- Hash mismatch, removing...
-- Using src='https://artifactory.kinovaapps.com:443/artifactory/generic-public/kortex/API/2.6.0/linux_aarch64_gcc_7.4.zip'
-- [download 0% complete]
-- [download 1% complete]
...
-- [download 100% complete]
-- verifying file...
       file='/robot_ws/kortex_ros2_ws/build/kortex_api/_deps/kinova_binary_api-subbuild/kinova_binary_api-populate-prefix/src/linux_aarch64_gcc_7.4.zip'
-- MD5 hash of
    /robot_ws/kortex_ros2_ws/build/kortex_api/_deps/kinova_binary_api-subbuild/kinova_binary_api-populate-prefix/src/linux_aarch64_gcc_7.4.zip
  does not match expected value
    expected: '64bd86e7ab8bda90ef1fc7d6a356e080'
      actual: 'd7771a7c106888717ace359164d40235'
-- Hash mismatch, removing...
-- Using src='https://artifactory.kinovaapps.com:443/artifactory/generic-public/kortex/API/2.6.0/linux_aarch64_gcc_7.4.zip'
-- [download 0% complete]
-- [download 1% complete]
...
-- [download 100% complete]
-- verifying file...
       file='/robot_ws/kortex_ros2_ws/build/kortex_api/_deps/kinova_binary_api-subbuild/kinova_binary_api-populate-prefix/src/linux_aarch64_gcc_7.4.zip'
-- MD5 hash of
    /robot_ws/kortex_ros2_ws/build/kortex_api/_deps/kinova_binary_api-subbuild/kinova_binary_api-populate-prefix/src/linux_aarch64_gcc_7.4.zip
  does not match expected value
    expected: '64bd86e7ab8bda90ef1fc7d6a356e080'
      actual: 'd7771a7c106888717ace359164d40235'
[... 44 lines cut ...]
gmake[2]: *** [CMakeFiles/kinova_binary_api-populate.dir/build.make:99: kinova_binary_api-populate-prefix/src/kinova_binary_api-populate-stamp/kinova_binary_api-populate-download] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/kinova_binary_api-populate.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2

CMake Error at /usr/share/cmake-3.22/Modules/FetchContent.cmake:1087 (message):
  Build step for kinova_binary_api failed: 2
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FetchContent.cmake:1216:EVAL:2 (__FetchContent_directPopulate)
  /usr/share/cmake-3.22/Modules/FetchContent.cmake:1216 (cmake_language)
  /usr/share/cmake-3.22/Modules/FetchContent.cmake:1259 (FetchContent_Populate)
  CMakeLists.txt:11 (FetchContent_MakeAvailable)

---
Failed   <<< kortex_api [8.99s, exited with code 1]
Aborted  <<< controller_manager_msgs [23.0s]      
Aborted  <<< ros_gz_interfaces [31.7s]            
Aborted  <<< control_msgs [39.4s]                 
Aborted  <<< joint_limits [41.1s]                 
Aborted  <<< ros_gz_sim [1min 1s]                                         

Summary: 3 packages finished [1min 2s]
  1 package failed: kortex_api
  5 packages aborted: control_msgs controller_manager_msgs joint_limits ros_gz_interfaces ros_gz_sim
  1 package had stderr output: kortex_api
  54 packages not processed```

martinleroux, 2024-06-18:
It seems that you did not update the MD 5 URL_HASH in the CMake file when you changed the URL for the API (see line 7).

You can find the proper hash by scrolling down  in the General panel of the Artifactory link. At the time of writing, the correct hash for the linux_aarch64_gcc_7.4 library is d7771a7c106888717ace359164d40235.

mehradmrt, 2024-06-19:
Correct. Changed the URL_Hash and the problem was resolved. Thanks!

closed 2024-06-19 by martinleroux