Kinovarobotics/ros2_kortex #226: Gen3 robotiq_gripper_controller failing to activate

keenan88, 2024-06-20:
Hello,

**Project Goal**

I am working on a minimal example for the 6 DOF gen3 arm. My original goal was to test the octomap with the kortex interface, but in the meantime I have run into a seperate issue. 

**Observed Behaviour**

When I launch the nodes in moveit_bringup.launch.py (attached below), I get the following output (attached below):
[moveit_bringup.launch.txt](https://github.com/user-attachments/files/15919328/moveit_bringup.launch.txt)
[minimal_example_output_1.txt](https://github.com/user-attachments/files/15919343/minimal_example_output_1.txt)
The control node does not see an interface for robotiq_85_left_knuckle_joint/position, and the robotiq gripper controller fails to activate. I can run the same launch file with use_fake_hardware = True, and the error is not present. However, I need to work on a physical arm. 

**Specs**

Gen3 arm has been updated the latest software available from Kinova. 
[image]
Running ROS2 humble from a docker container

If you want to run my minimal example, the repo is here:
https://github.com/keenan88/kinova_gen3_octomap_minimal#js-repo-pjax-container
To reproduce the issue:
1. Clone the repo
2. Ensure gen3 is connected to your computer via ethernet.
3. From the root, run docker compose up.
4. Run docker exec -it humble-humble-moveit bash to enter the container.
5. Within the container, run ros2 launch antworker_moveit_bringup moveit_bringup.launch.py to start the minimal example launch file.
6. The output should be the same as in the output text file above.

I have managed to reproduce the same issues using the ros2_kortex branch on main and tag 0.2.2. 

Can anyone shed some light on what is happening?

Thank you,

Keenan

aalmrad, 2024-06-21:
Hello, 
Thank you for reaching out. We are currently investigating the gripper control to identify the discrepancy between the simulation and hardware implementation.

aalmrad, 2024-06-28:
The issue was successfully reproduced on my end. The robotiq_85_left_knuckle_joint is properly declared as a position joint under the gripper controller and the joint is properly included in the robot description yet for a certain reason that i couldn't identify yet, the joint is not recognized. I am actively working on finding a solution.

closed 2024-10-07 by aalmrad