Kinovarobotics/ros2_kortex #227: Kortex ros2 control node crashes when using octomap with gen3 arm

keenan88, 2024-06-21:
Hello,

**Project Goal**

I am trying to integrate the kortex control nodes with Moveit2's octomap functionality. 

**Steps Taken**

I have added a 3d_sensors.yaml file to the moveit configuration in kinova_gen3_6dof_robotiq_2f_85_moveit_config/config. 

**Observed Behaviour**

When I provide pointcloud data from a realsense to the topic specified in the 3d_sensors.yaml file, I see an octomap populate in RVIZ, with voxels. After a few seconds, the ros2 control node crashes (output linked below). This is the behaviour on my Lenovo Legion 15ACH6, using USB 3.2 ports with 5Gb/s. I also have the realsense depth & IR at low res/fps (848x480x5), the octomap refreshing at just 1Hz, and have tried reducing the gen3 refresh rate to 500, 100, and 10Hz. I can run the realsense without feeding its depth information forward to the octomap topic, and no crash occurs.

Using the exact same code, this behaviour does not occur when I use an Intel Nuc10FNH. In USBViewer, the usb ports on the Nuc also show up as Generation 3.2 at 5Gb/s. 

**Resources**

Launch file: [moveit_bringup.launch.txt](https://github.com/user-attachments/files/15930116/moveit_bringup.launch.txt)

RVIZ before realsense started & octomap populated:
[image]

RVIZ after realsense started & octomap populated:
[image]

Terminal output (I put some line breaks between before/after starting the realsense): 
[octomap_kortex_output.txt](https://github.com/user-attachments/files/15930120/octomap_kortex_output.txt)

My project repo: https://github.com/keenan88/isaacsim_ros2_greenhouse/tree/voxels_working

Are there any explanations for this behaviour?

Thank you,

Keenan

aalmrad, 2024-10-04:
Hello @keenan88 ,

From your description it seems that your Lenevo PC is facing real-time processing issues and is having a hard time to handle both the sensor's data streams and the robotic arm feedback simultaneously. Can you please provide more info about the following:
1- Can you please provide the specs of both your Lenovo PC and the Intel Nuc10FNH kit? 
2- I suppose you are using CycloneDDS, did you try to modify the DDS settings or use a different DDS type?
3- Did you monitor the CPU load and the network latency and jitter on your PC when the timeout happens?

Best,
Abed

aalmrad, 2026-04-03:
Claude has submitted a fix attempt: https://github.com/Kinovarobotics/ros2_kortex/pull/377

    Please review the proposed modifications and test them on your end, then provide feedback.
    A member of Kinova's team will follow up on this issue and take the appropriate next steps.