Kinovarobotics/ros2_kortex #230: Gen3 Lite gripper control issue

PrwTsrt, 2024-07-24:
I can't control the gripper with MoveIt when I execute the gripper plan; it always fails. However, the arm operates without any issues.

These are my configs and launch file.

**robot.launch.py**
```py
import os

from launch import LaunchDescription
from launch.substitutions import LaunchConfiguration
from launch.actions import (
    DeclareLaunchArgument,
    OpaqueFunction,
    RegisterEventHandler,
)
from launch.event_handlers import OnProcessExit
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory
from moveit_configs_utils import MoveItConfigsBuilder

def launch_setup(context, *args, **kwargs):
    # Initialize Arguments
    robot_ip = LaunchConfiguration("robot_ip")
    use_fake_hardware = LaunchConfiguration("use_fake_hardware")
    gripper_max_velocity = LaunchConfiguration("gripper_max_velocity")
    gripper_max_force = LaunchConfiguration("gripper_max_force")
    launch_rviz = LaunchConfiguration("launch_rviz")
    use_sim_time = LaunchConfiguration("use_sim_time")
    use_internal_bus_gripper_comm = LaunchConfiguration("use_internal_bus_gripper_comm")

    launch_arguments = {
        "robot_ip": robot_ip,
        "use_fake_hardware": use_fake_hardware,
        "gripper": "gen3_lite_2f",
        "gripper_joint_name": "right_finger_bottom_joint",
        "dof": "6",
        "gripper_max_velocity": gripper_max_velocity,
        "gripper_max_force": gripper_max_force,
        "use_internal_bus_gripper_comm": use_internal_bus_gripper_comm,
    }

    moveit_config = (
        MoveItConfigsBuilder("gen3_lite_gen3_lite_2f", package_name="kinova_gen3_lite_moveit_config")
        .robot_description(mappings=launch_arguments)
        .trajectory_execution(file_path="config/moveit_controllers.yaml")
        .planning_scene_monitor(
            publish_robot_description=True, publish_robot_description_semantic=True
        )
        .planning_pipelines(pipelines=["ompl", "pilz_industrial_motion_planner"])
        .to_moveit_configs()
    )

    moveit_config.moveit_cpp.update({"use_sim_time": use_sim_time.perform(context) == "true"})

    move_group_node = Node(
        package="moveit_ros_move_group",
        executable="move_group",
        output="screen",
        parameters=[
            moveit_config.to_dict(),
        ],
    )

    # Static TF
    static_tf = Node(
        package="tf2_ros",
        executable="static_transform_publisher",
        name="static_transform_publisher",
        output="log",
[... 226 lines cut ...]
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.0
      goal_time: 0.0
      
gen3_lite_hand_controller:
  ros__parameters:
    joint: right_finger_bottom_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity

twist_controller:
  ros__parameters:
    joint: tcp
    interface_names:
      - twist.linear.x
      - twist.linear.y
      - twist.linear.z
      - twist.angular.x
      - twist.angular.y
      - twist.angular.z
```

martinleroux, 2024-07-24:
Hi @PrwTsrt ,
I believe we are already working on a fix for this. The person who is handling it is currently travelling to a tradeshow. I will forward him this issue to ensure he sees it upon his return at the end of the week.

Cheers

aalmrad, 2024-07-26:
Hello @PrwTsrt ,
Thank you for reaching out. After checking your files, I can see that a separate group was created for the right_finger_bottom_joint with a separate controller inside Moveit2. However in the case of the Gen3-Lite, the gripper is part of the robotic arm and should be treated as such, therefore the right_finger_bottom_joint should be added to the group containing the rest of the robotic arm's joints. I have created a moveit2 package for the Gen3-Lite that addresses this issue and you can find it in the following pull request: https://github.com/Kinovarobotics/ros2_kortex/pull/231

PrwTsrt, 2024-07-31:
Hello @rahulpatel,

Thank you for your prompt response and for addressing the issue. I integrated the moveit2 package from pull request #231, and everything is working perfectly now.

I appreciate your support and the quick resolution. If I encounter any further issues or have additional questions, I'll be sure to reach out.

Thanks again!

PrwTsrt, 2024-07-31:
Hello @rahulpatel,

Thank you for your prompt response and for addressing the issue. I integrated the moveit2 package from pull request https://github.com/Kinovarobotics/ros2_kortex/pull/231, and everything is working perfectly now.

I appreciate your support and the quick resolution. If I encounter any further issues or have additional questions, I'll be sure to reach out.

Thanks again!

PrwTsrt, 2024-08-07:
After rebuilding my workspace and cloning the Git repository, I discovered that I am unable to control the arm without the gripper. Is it possible to separate the control of the arm and the gripper using MoveIt?

[image]

aalmrad, 2024-08-08:
Hello @PrwTsrt , 

Using the setup assistant, you can customize the moveit package that i have created for the Gen3 Lite to your liking. For instance, you can put the gripper's joint under a separate control group to plan and execute its motion separately from the arm. Just a quick note, my name tag is @aalmrad . I'm here to help with any other questions or solutions you might need.

PrwTsrt, 2024-08-09:
Hello @aalmrad, I apologize for the misunderstanding earlier. I attempted to separate the gripper into its own control group. However, when I set the controller type to FollowJointTrajectory, I encountered an error 
```
[ros2_control_node-1] During ros2_control interface configuration, degrees of freedom is not valid; it should be positive. Actual DOF is 0
```
I also changed the controller type to GripperCommand, which allowed me to control the arm without issues. Unfortunately, when I try to execute the gripper plan, it consistently fails.

**.srdf file**
```xml
<group name="arm">
        <joint name="base_joint"/>
        <joint name="joint_1"/>
        <joint name="joint_2"/>
        <joint name="joint_3"/>
        <joint name="joint_4"/>
        <joint name="joint_5"/>
        <joint name="joint_6"/>
    </group>
    <group name="hand">
        <joint name="right_finger_tip_joint"/>
        <joint name="right_finger_bottom_joint"/>
        <joint name="left_finger_tip_joint"/>
        <joint name="left_finger_bottom_joint"/>
        <joint name="gripper_base_joint"/>
    </group>
    <!--END EFFECTOR: Purpose: Represent information about an end effector.-->
    <end_effector name="hand" parent_link="end_effector_link" group="hand"/>
    <!--PASSIVE JOINT: Purpose: this element is used to mark joints that are not actuated-->
    <passive_joint name="left_finger_bottom_joint"/>
    <passive_joint name="left_finger_tip_joint"/>
    <passive_joint name="right_finger_tip_joint"/>
 ```

   **ros2_controllers.yaml**

```yaml
controller_manager:
  ros__parameters:
    update_rate: 1000  # Hz

    arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    hand_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

arm_controller:
  ros__parameters:
    joints:
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
hand_controller:
  ros__parameters:
    joint: right_finger_bottom_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
```
   **moveit_controllers.yaml**
```yaml
moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - arm_controller
    - hand_controller

  arm_controller:
    type: FollowJointTrajectory
    action_ns: follow_joint_trajectory
    default: true
    joints:
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6
    action_ns: follow_joint_trajectory
    default: true
  hand_controller:
    type: FollowJointTrajectory
    action_ns: follow_joint_trajectory
    default: true
    joints:
      - right_finger_bottom_joint
    action_ns: follow_joint_trajectory
    default: true
```

aalmrad, 2024-10-04:
Hello, the error you are getting suggests that you have issues with your kinematic chain definition. Upon inspecting your .srdf file I can see some problems:
1- The end_effector should be part of the "arm" group and not the "hand" group, since the `end_effector_link` is defined in the arm`s URDF and not the gripper URDF
2- The `right_finger_bottom_joint` is supposed to be the only active joint in the hand/gripper and the rest of the joints are supposed to be passive, therefore only this joint should be included in the "hand" group otherwise this might raise issues since for a parallel gripper the kinematic chain is not serial and this causes confusion for moveit and ros2
 
 Please adress these issues and keep me posted regarding the situation.

closed 2024-11-06 by aalmrad