Kinovarobotics/ros2_kortex #239: Align depth and color images with ros2_kortex_vision

Eric-nguyen1402, 2024-09-26:
Hi 
I am using the Gen 3 Lite Robot 7DoF with the Realsense D410 camera. For my project, I need RGBD images from the camera, which I access using the ros2_kortex_vision package. Based on the GitHub documentation, I assumed that the /camera/depth_register/image_rect topic is aligned with the /camera/color/image_raw topic. However, when I tested this using plotjuggler, it showed a significant time difference between these two topics, even after alignment.

I believe this time gap is why the `message_filters.ApproximateTimeSynchronizer` or `message_filters.TimeSynchronizer` functions fail to synchronize the messages and trigger the callback function. Interestingly, in some tests, it worked briefly, which makes me think the issue isn’t with my code, but rather with these two topics being out of sync.

Do you have any suggestions on how to fix this issue?
[image]

martinleroux, 2024-09-27:
Given that the ros2_kortex_vision package is not maintained by Kinova, there is little I can do about this. I will say that 0.1 second of delay between the two frames is within expected values for a Gen3. If you just open the rtsp stream (e.g. with VLC with rtsp://192.168.1.10/leftir), you will see that the video input itself has a lot of lag.

I suggest submitting your issue to the ros2_kortex_vision repo.

Eric-nguyen1402, 2024-09-27:
Thanks for your suggestion but it looks like the ros2_kortex_vision repo has disable the issues. That's why I open the new issue here to ask for help.

martinleroux, 2024-09-27:
Then the only thing I can add is that most of the customers I heard from who are using vision are mapping RGB to depth on captures while the robot is immobile - in these cases, the delay does not matter. I'm sure some of them must  map the streams, but I don't know how they do it. 

I will leave this issue open, just in case one such user comes across it and can give you an answer.

peterminh227, 2024-09-29:
Dear @martinleroux, 

I had the same issues with these two topics, /camera/depth_register/image_rect and /camera/color/image_raw. Streaming two topics from the robot is too unstable. This makes me wonder if it does not make sense to use the attached camera with KINOVA. I regret buying the KINOVA vision version when this utility in ROS 2 is so unstable. 

I hope KINOVA can step in and implement a stable ros2 vision version

Best regards,

martinleroux, 2024-09-30:
Hi @peterminh227 ,
I understand your situation. I have already added taking a second look at the vision driver to my team's todo list. As I highlighted earlier, the current driver was not developed and is not maintained by Kinova.

Out of curiosity, can you tell me what is the output of htop while you are trying to operate vision? An unrelated investigation on our side recently revealed that ROS2 requires much more processing power under certain circumstances (the specifics of which we have yet to pin down) than what we expected. We've had PCs that could run ROS1 applications just fine but simply could not handle ROS2.

aalmrad, 2024-10-03:
Hello @Eric-nguyen1402 ,

I have used the [ros2_kortex_vision](https://github.com/PickNikRobotics/ros2_kortex_vision) github repository with a Gen3 on my end and here are my findings:
First, i have created a node to subscribe to both color and depth streams (i.e. "/camera/color/image_raw" and "/camera/depth_registered/image_rect" topics) and calculate the timestamp differences among the received messages and report the average of each 100 readings and here are the results:
[image]

Also, here is a plot showing the differential readings in both seconds and nanoseconds using  plotJuggler:
[image]

As you can see, the values are mostly well below 0.1 seconds

Second, i have created a synchronization node using a Synchronizer with a an approximate time policy (slop= 0.1 and queue size = 10) and it is working well although i am sure there will be some lost out of sync messages and here's the source code for the node:
```
#include <rclcpp/rclcpp.hpp>
#include <sensor_msgs/msg/image.hpp>
#include <message_filters/subscriber.h>
#include <message_filters/sync_policies/approximate_time.h>
#include <message_filters/synchronizer.h>

using std::placeholders::_1;
using std::placeholders::_2;

class ImageSyncNode : public rclcpp::Node
{
public:
    ImageSyncNode()
    : Node("image_sync_node")
    {
        // Create subscribers for color and depth topics
        color_sub_ = std::make_shared<message_filters::Subscriber<sensor_msgs::msg::Image>>(
            this, "/camera/color/image_raw");

        depth_sub_ = std::make_shared<message_filters::Subscriber<sensor_msgs::msg::Image>>(
            this, "/camera/depth_registered/image_rect");

        // Define the synchronization policy with a specific slop value
        typedef message_filters::sync_policies::ApproximateTime<sensor_msgs::msg::Image, sensor_msgs::msg::Image> SyncPolicy;

        // Set the slop value (in seconds) and queue size for synchronization
        const double slop = 0.1;  // Change this value to adjust the tolerance for time difference
        const int queue_size = 10; // The number of messages to buffer

        // Create the synchronizer using the defined policy
        sync_ = std::make_shared<message_filters::Synchronizer<SyncPolicy>>(SyncPolicy(queue_size), *color_sub_, *depth_sub_);
        // Set the slop value separately
        sync_->setMaxIntervalDuration(rclcpp::Duration::from_seconds(slop)); 
        // Register the callback function for synchronized messages
        sync_->registerCallback(std::bind(&ImageSyncNode::callback, this, _1, _2));
    }

private:
    // Callback function for synchronized messages
    void callback(const sensor_msgs::msg::Image::ConstSharedPtr & color_msg, const sensor_msgs::msg::Image::ConstSharedPtr & depth_msg)
    {
        RCLCPP_INFO(this->get_logger(), "Synchronized pair received");
        RCLCPP_INFO(this->get_logger(), "Color timestamp: %u.%u", color_msg->header.stamp.sec, color_msg->header.stamp.nanosec);
        RCLCPP_INFO(this->get_logger(), "Depth timestamp: %u.%u", depth_msg->header.stamp.sec, depth_msg->header.stamp.nanosec);

        // Here, you can process the synchronized color and depth images
    }

    // Subscribers and synchronizer
    std::shared_ptr<message_filters::Subscriber<sensor_msgs::msg::Image>> color_sub_;
    std::shared_ptr<message_filters::Subscriber<sensor_msgs::msg::Image>> depth_sub_;
    std::shared_ptr<message_filters::Synchronizer<message_filters::sync_policies::ApproximateTime<sensor_msgs::msg::Image, sensor_msgs::msg::Image>>> sync_;
};

int main(int argc, char * argv[])
{
    // Initialize ROS2
    rclcpp::init(argc, argv);
    
    // Create an instance of the ImageSyncNode
    auto node = std::make_shared<ImageSyncNode>();
    
    // Spin the node
    rclcpp::spin(node);
    
    // Shutdown ROS2
    rclcpp::shutdown();
    return 0;
}

```
I hope this helps.

Cheers,
Abed

Eric-nguyen1402, 2024-10-11:
Dear @aalmrad  
Thank you for your findings. I actually tried a similar approach, but instead of creating a new node, I opted to use the message_filters library, specifically the `message_filters.ApproximateTimeSynchronizer` or `message_filters.TimeSynchronizer`. These functions only output when they have received messages from all specified sources with matching timestamps, eliminating the need to manually compute or predict the timestamps which is the same method like you.

I also used message_filters.ApproximateTimeSynchronizer with a slop of 0.1 and a queue size similar to yours, but it only worked in a few cases and was too unstable for the project. Therefore, I believe the core issue may not lie in how we synchronize the topics.

Best regards,
Eric

aalmrad, 2026-04-03:
As part of Kinova’s beta testing: Claude is analyzing this issue and will attempt an automated fix. A pull request will be opened if changes are made.