Kinovarobotics/ros2_kortex #245: Gen 3 Lite MoveIt Limp Gripper

htchr, 2024-10-16:
Hello,

I'm trying to control the gripper using the included MoveIt configuration for the gen3 lite robot arm.
However, it seems that only one side can be controlled, and the other side hangs limp (see screenshot). 
Therefore it cannot actually grip anything in simulation; the object always falls away.
Is there anything I can do to fix this issue? 

Thank you,

[image]

ssapsu, 2024-10-16:
hi

i think the problem is it's because symmetric
one side is clockwise and the other size is anti clockwise

ssapsu, 2024-10-16:
like mirror

ssapsu, 2024-10-16:
I had kind of that problem in isaac sim too

htchr, 2024-10-16:
Hello @ssapsu were you able to fix the issue? I'm using the default Gazebo and MoveIt configurations.

ssapsu, 2024-10-16:
I think it's because of limitation of implement specific joint like in your case and my case which was robotiq 2f 85. 

Becasue it opens and closes symmetric with single motor but i'm afraid they implemented separately

In my case i filtered ros2 topic to only hear left side and i gave a command on the left side

Maybe you can change your gripper's config file so you can correct right

closed 2024-11-06 by htchr