Kinovarobotics/ros2_kortex #246: Gen 3 Lite MoveIt Tutorial / Examples

htchr, 2024-10-16:
Hello,

I am trying to use the Gen3 Lite robot with MoveIt2. 
I am able to launch it using Gazebo and Rviz and move the arm around with the interactive marker.
However, I haven't been able to figure out how to move the arm programmatically using python.
Are there any examples or tutorials I should reference?
MoveIt2 doesn't seem to have many python examples, so I thought I would ask.

Thank you,

aalmrad, 2024-10-21:
Hello @htchr ,

Please refer to the [moveit tutorials](https://moveit.picknik.ai/main/doc/examples/examples.html#using-moveit-directly-through-the-python-api) from picknik and the [pymoveit2 repository examples](https://github.com/AndrejOrsula/pymoveit2/tree/master/examples) might be helpful as well.

Best,
Abed

htchr, 2024-12-10:
For anyone with a similar question, here is an example I wrote with pymoveit2 that works fairly well: https://github.com/GIXLabs/TECHIN516/blob/master/lab7/gen3lite_pymoveit2.py
Thanks,

closed 2024-12-10 by htchr