Kinovarobotics/ros2_kortex #248: timeout detected: BaseCyclicClient::RefreshFeedback

MiguelGarciaUVa, 2024-10-30:
Hi,

I'm encountering an issue where the execution of my node stops intermittently due to an error. Here’s the message I receive:

[ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-1]   what():  timeout detected: BaseCyclicClient::RefreshFeedback
[ros2_control_node-1]
[ERROR] [ros2_control_node-1]: process has died [pid 2240, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_k4apqsns --params-file /opt/ros/humble/share/kinova_gen3_7dof_robotiq_2f_85_moveit_config/config/ros2_controllers.yaml'].

Another error I have is this one:

[ros2_control_node-1] [ERROR] [1730297841.669743775] [KortexMultiInterfaceHardware]: Runtime error: timeout detected: BaseCyclicClient::Refresh
[ros2_control_node-1] 
[ros2_control_node-1] [ERROR] [1730297842.610749945] [KortexMultiInterfaceHardware]: Future error: std::future_error: Future already retrieved

The last error occurs more frequently when I use the camera with the command ros2 launch kinova_vision kinova_vision.launch.py.
CPU Usage is not a problem.
Does anyone have insights into what could be causing this timeout error?

Thanks!

Melih199, 2025-01-22:
I am facing with the same issue. Did you find any solution for it? @MiguelGarciaUVa 

Thanks!

MiguelGarciaUVa, 2025-02-11:
No, I didn´t find a solution but I think it depends on the computational load of the computer. It happens more when I have the computer close to its maximum performance. 
I have the arm on a mobile platform and the control is on a NUC. I bought a better NUC but it still happens..

aalmrad, 2025-02-11:
Hello,

We have noticed this problem on our side when running computationally intensive applications and we are working towards finding a solution.

Best,
Abed

martinleroux, 2025-02-11:
To add on @aalmrad 's comment, we did indeed stumble unto the same symptoms on our side. As @MiguelGarciaUVa has highlighted, we have also noticed it seems to be related to the computational load on the PC. Running the code on a better PC did improve things a lot, but I understand this is not necessarily a viable solution for everyone. Our next best recommendation is to use the RT patch for Linux on your PC to stabilize the real-time connection.

We are still looking for better solutions that could solve the issue entirely. If we end up finding anything, we'll make sure to update this issue. Although I admittedly can't promise we will find a universal solution since the occurrence is linked to things happening on the PC outside of the driver itself.

DavidG-Develop, 2025-08-20:
@martinleroux @aalmrad @MiguelGarciaUVa  Have you found any type of patch or solution to this issue? I am running with an Intel i9 14 gen + 32gb of ram, and I am encountering this issue very often... The only partial solution so far was using the compressed topics of the vision with very low resolution and fps... This seems to more or less function but with that the camera data is almost not usable, and it still crashes as soon as I run anything a bit more power hungry... My only solution atm would be running of 2 pcs (1 jetson + 1 pc) but that seems like a huge overkill to control a single arm...