Kinovarobotics/ros2_kortex #260: Inconsistencies with ROS Humble

fabioscap, 2025-01-15:
Hello! We have a Gen3 Lite. I am trying to get the drivers working on ROS humble. I am installing the binaries with `apt`. 
Running `ros2 launch kortex_description view_robot.launch.py robot_type:=gen3_lite dof:=6` returns, notably:

    Captured stderr output: error: No such file or directory: /opt/ros/humble/share/kortex_description/arms/gen3_lite/7dof/urdf/gen3_lite_macro.xacro

It looks like in my local installation view_robot.py does not catch the `dof` argument which needs to be passed to `kinova.urdf.xacro`.  Similar errors appear if I intend to launch kortex_bringup, as `gen3_lite.launch.py` does not seem to be existing. Therefore I cannot follow the README.md instructions. 

Would it be possible to have more up to date Humble packages? Many thanks

fabioscap, 2025-01-15:
Hello @aalmrad,

As stated I am using the apt packages `ros-humble-kortex-bringup` and `ros-humble-kortex-description`, which are not up to date with `Humble_Gazebo_Classic_Support`. Nevertheless I try building directly this branch, Thanks for the quick get back!

Best,
Fabio

aalmrad, 2025-01-16:
Hello @fabioscap,

You are right. Please let me know if you face any issues with building from source.

Best,
Abed

fabioscap, 2025-01-17:
Hello,

I was able to compile and run `Humble_Gazebo_Classic_Support`.

Best,
Fabio

closed 2025-01-17 by fabioscap