Kinovarobotics/ros2_kortex #261: Moveit Servo setup

AGregg, 2025-01-15:
Hello, 
I am getting very similar issues as in the previous issue here marked as resolved: https://github.com/Kinovarobotics/ros2_kortex/issues/168

I updated my project to Jazzy as that seemed to be the expected response, but I still have the same issues.  

When I have a low update frequency, I get choppy motion.  At higher frequencies the robot moves slow but smoothly, however there is an effect where the robot droops down with gravity over time, even with no twist command input.

I have tested with position controllers, and joint trajectory controllers.  Tested different `publish_period` and `max_expected_latency` values, tested with smoothing filters, low latency mode on/off, etc.  I have also tried doing a feed forward approach where I publish a "fake" joint state topic that initializes to what the actual joint states are but then follows what positions were last commanded to the robot, as I suspected that the drooping came from commanding back out very minute changes in the joint states from gravity.  

Is there any guidance on how to configure moveit servo with the gen 3 kinova arm, such as what interfaces to use for commands in and out, whether to publish velocity or position, what moveit servo parameters such as publish period and max expected latency to use, what update rates to use in the ros2_controllers.yaml, etc.

aalmrad, 2025-01-20:
Hello @AGregg,

Your request has been well received. I will get back to you with solutions very soon.

Best,
Abed

sea-bass, 2025-01-29:
I may point you to this configuration from PickNik:
https://github.com/PickNikRobotics/moveit_studio_kortex_ws/blob/main/src/kinova_gen3_base_config/config/moveit/kinova_gen3_servo.yaml

Maybe that's useful / works? Haven't ever used it myself.