Kinovarobotics/ros2_kortex #262: Orientation control with twist_controller not working

nbuttner-bdai, 2025-01-17:
## Description

When using a kinova gen3 with the twist_controller enabled, orientation control does not work (publishing a non zero angular velocity results in no motion of the arm).

## Version

ROS distribution : Humble

Branch and commit you are using : [2bf1526cf26699f025832a60ff6f2c79ffa0d1c7](https://github.com/Kinovarobotics/ros2_kortex/tree/2bf1526cf26699f025832a60ff6f2c79ffa0d1c7)

## Steps to reproduce

1. `ros2 launch kortex_bringup gen3.launch.py  robot_ip:=192.168.1.10 robot_controller:=twist_controller` 
2. `ros2 topic pub /twist_controller/commands geometry_msgs/msg/Twist "{angular: {x: 0.1}}"`
3. The arm does not move.

## Expected behavior

Non zero angular velocities in the spatial twist message result in motion of the arm.

## Any other information

The linear velocities (XYZ), all work as expected. If non zero linear and angular velocities are sent, only the linear velocities are applied. The arm works properly when being commanded either via the web UI, or with the xbox controller.

aalmrad, 2025-01-21:
Hello @nbuttner-bdai,

Both the linear and angular twist commands are working well on my side using the latest commit on the  `Humble_Gazebo_Classic_Support` branch.

Please verify if the angular interfaces are available and claimed using the following terminal command:
`ros2 service call /controller_manager/list_hardware_interfaces controller_manager_msgs/srv/ListHardwareInterfaces`

I will be waiting for your feedback.

Best,
Abed

nbuttner-bdai, 2025-01-22:
Hi @aalmrad ,

Thanks for your reply! Yes, I do see the angular interfaces listed as available and claimed. I believe switching from main to `Humble_Gazebo_Classic_Support` may have fixed the issue, as it now works! Thank you for your help!