Kinovarobotics/ros2_kortex #264: Joint 'robotiq_85_left_knuckle_joint' not found in URDF

Sayantan405, 2025-02-08:
**Sytem info:-** Followed ateps on README to set the kinova with ros2 , the system I'm using is a intel process with Graphics.
**what's the goal:-** to teleop the kinova arm with keyboard in ros2 env.
**am'I using conda env?** NO ros2 installed system wide.
**Issue:-**  ros2 control node not able to fine robotiq_85_left_knuckle_joint in urdf.

```
[robot_state_publisher-2] [INFO] [1739053816.016799705] [robot_state_publisher]: Robot initialized
[ros2_control_node-1] [INFO] [1739053816.026524148] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-1] [INFO] [1739053816.027713677] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-1] [INFO] [1739053816.027722030] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1739053816.027808004] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-1] [INFO] [1739053816.036833527] [controller_manager]: Received robot description from topic.
[ros2_control_node-1] Error parsing the limits for the interface: position from the tags [min: '{-2*pi}' and max: '{2*pi}'] within ros2_control tag inside the URDF. Skipping it
[ros2_control_node-1] Error parsing the limits for the interface: position from the tags [min: '{-2*pi}' and max: '{2*pi}'] within ros2_control tag inside the URDF. Skipping it
[ros2_control_node-1] Error parsing the limits for the interface: position from the tags [min: '{-2*pi}' and max: '{2*pi}'] within ros2_control tag inside the URDF. Skipping it
[ros2_control_node-1] Error parsing the limits for the interface: position from the tags [min: '{-2*pi}' and max: '{2*pi}'] within ros2_control tag inside the URDF. Skipping it
[ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-1]   what():  Joint 'robotiq_85_left_knuckle_joint' not found in URDF
[ros2_control_node-1] Stack trace (most recent call last) in thread 81244:
[ros2_control_node-1] #18   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-1] #17   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 78, in clone3 [0x707eb9129c3b]
[ros2_control_node-1] #16   Source "./nptl/pthread_create.c", line 447, in start_thread [0x707eb909caa3]
[ros2_control_node-1] #15   Source "../../../../../src/libstdc++-v3/src/c++11/thread.cc", line 104, in execute_native_thread_routine [0x707eb94ecdb3]
```

aalmrad, 2025-02-11:
Hello,

Can you please specify the default value for the "gripper_joint_name" argument to be an empty string in the gen3.launch.py file?

Best,
Abed

aalmrad, 2025-05-14:
@Sayantan405 I am not facing this issue on my end. Can you please share your modified launch file?

aalmrad, 2026-04-03:
Claude has submitted a fix attempt: https://github.com/Kinovarobotics/ros2_kortex/pull/373

    Please review the proposed modifications and test them on your end, then provide feedback.
    A member of Kinova's team will follow up on this issue and take the appropriate next steps.