Kinovarobotics/ros2_kortex #276: Installing ros_jazzy_kortex_bringup

otokatli, 2025-04-04:
Hi,

I would like to install ros_jazzy_kortex_bringup using the apt package manager inside my docker container which is running osrf/ros:jazzy-desktop-full image. Unfortunately, apt is unable to locate the package in the known repositories. My ROS skills are rusted, so I might be missing a step here but I cannot figure out how to solve this problem. One option is compiling the ros2_kortex myself using the main branch of the git repository. However, I would prefer the binary installation. How can I install ros_jazzy_kortex_bringup?

aalmrad, 2025-04-07:
@otokatli ,

The latest binary packages we have released were for ROS2 Humble. We will release binary packages for ROS2 Jazzy but it will take some time as this is not our priority at the moment. Thank you for your understanding.

Best,
Abed

otokatli, 2025-04-08:
Thank you very much for your response. I’m looking forward to downloading the binary files. 😄

closed 2025-04-14 by otokatli