Kinovarobotics/ros2_kortex #281: Trajectories being recorded with extremely small values

Abhishek8857, 2025-05-05:
I am trying to record the trajectories the robot takes when I exectue joint position based movements. However, the recorded trajectories have extremely minute values that the robot maybe is not able to recognise when I try to replay. 

Executing Motion + Trajctory Recording
``` 
# include <memory>
# include <rclcpp/rclcpp.hpp>
# include <geometry_msgs/msg/point.hpp>
# include <moveit/move_group_interface/move_group_interface.h>
# include <std_msgs/msg/float64_multi_array.hpp>
# include <std_msgs/msg/bool.hpp>
# include <thread>
# include <chrono>
# include <yaml-cpp/yaml.h>
# include <fstream>
# include <iomanip>
# include <filesystem>

using std::placeholders::_1;

class AgentSubscriber : public rclcpp::Node
{
  public:
    // Initialise the Node
    AgentSubscriber () : Node ("agent_subscriber")
    {
      // Subscribe to recieve Coordinates
      RCLCPP_INFO(this->get_logger(), "Agent subscriber node initialised. Waiting for coordinates...");
      subscription = this->create_subscription<std_msgs::msg::Float64MultiArray>(
        "published_coordinates", 10, std::bind(&AgentSubscriber::agent_callback, this, _1));
    }

  private:  
    void agent_callback(const std_msgs::msg::Float64MultiArray::SharedPtr msg) 
    {
      // DEBUG: Print out the recieved Coordinates
      RCLCPP_INFO(this->get_logger(), "Recieved Cartesian Coordinates..");
      for (size_t i {0}; i < msg->data.size(); i++)
      {
        RCLCPP_INFO(this->get_logger(), " -[%zu]: %f", i, msg->data[i]);
      }

      // Lazy initialization of MoveGroupInterface
      if (!move_group_manipulator && !move_group_gripper)
      {
        move_group_manipulator = std::make_unique<moveit::planning_interface::MoveGroupInterface>(shared_from_this(), "manipulator");
        move_group_gripper = std::make_unique<moveit::planning_interface::MoveGroupInterface>(shared_from_this(), "gripper");
      } 

      // Check if we need to execute Joint Target or Pose Target
      if (msg->data[0] == 0.0)
      {
        // Parse the Coordinates
        std::map<std::string, double> joint_values = 
        {
          {"joint_1", msg->data[1]},
          {"joint_2", msg->data[2]},
          {"joint_3", msg->data[3]},
          {"joint_4", msg->data[4]},
          {"joint_5", msg->data[5]},
          {"joint_6", msg->data[6]},
          {"joint_7", msg->data[7]}
        };

        // Plan and execute
        RCLCPP_INFO(this->get_logger(), "Planning the target pose ...");
        move_group_manipulator->setJointValueTarget(joint_values);

        // Verify if the motion was successful
        // bool success = (move_group_manipulator->move() == moveit::core::MoveItErrorCode::SUCCESS);
        moveit::planning_interface::MoveGroupInterface::Plan plan;
[... 350 lines cut ...]
  - positions: [2.4436269677025023e-06, -0.02450712179177136, 0.1003687046943901, -0.068232492982692655, 0.00019112615066999762, -0.038438744397493729, 0.049651706735082395]
    velocities: [3.3995022107660062e-06, -0.034093589492903836, 0.13962999999999964, -0.094923044231593631, 0.00026588909859213625, -0.053474854503348226, 0.069073999037142439]
    accelerations: [-1.9928778576172264e-16, 1.9590786491520383e-12, -8.4893408129921659e-12, 5.5507228392641085e-12, -1.5305301946500299e-14, 3.1018745278240606e-12, -4.244670406496083e-12]
    time_from_start: 0.80000000000000004
  - positions: [2.7835771887790939e-06, -0.027916480741061653, 0.1143317046943897, -0.077724797405851767, 0.00021771506052921055, -0.043786229847828408, 0.056559106638796451]
    velocities: [3.3995022107658719e-06, -0.03409358949290249, 0.13962999999999415, -0.094923044231589884, 0.00026588909859212578, -0.053474854503346117, 0.069073999037139719]
    accelerations: [-1.9349248109138003e-16, 1.937855332051754e-12, -7.9395626225809725e-12, 5.3996551485325566e-12, -1.5066002088539501e-14, 3.0384819041394005e-12, -3.9321530524156947e-12]
    time_from_start: 0.90000000000000002
  - positions: [3.1235274098556859e-06, -0.031325839690351943, 0.12829470469438931, -0.087217101829010879, 0.0002443039703884235, -0.049133715298163093, 0.063466506542510515]
    velocities: [3.3995022107662539e-06, -0.03409358949290632, 0.13963000000000983, -0.094923044231600556, 0.00026588909859215565, -0.053474854503352126, 0.069073999037147477]
    accelerations: [5.5690952818201531e-16, -5.5876891602973662e-12, 2.2867537141563788e-11, -1.5568012573776479e-11, 4.3653821564823173e-14, -8.769119115726792e-12, 1.1304573445688314e-11]
    time_from_start: 1
  - positions: [3.4634776309322778e-06, -0.034735198639642244, 0.14225770469438892, -0.096709406252170005, 0.00027089288024763645, -0.054481200748497778, 0.070373906446224585]
    velocities: [3.3995022107659973e-06, -0.034093589492903746, 0.13962999999999928, -0.094923044231593395, 0.0002658890985921356, -0.053474854503348088, 0.069073999037142272]
    accelerations: [-1.980513411952634e-16, 1.9785812037458508e-12, -8.2308978075827388e-12, 5.3817408741887145e-12, -1.4839359028093882e-14, 3.0865866778435272e-12, -3.9571624074917015e-12]
    time_from_start: 1.1000000000000001
  - positions: [3.8034278520088694e-06, -0.038144557588932537, 0.15622070469438853, -0.10620171067532912, 0.00029748179010684941, -0.05982868619883247, 0.077281306349938655]
    velocities: [3.399502210765863e-06, -0.0340935894929024, 0.13962999999999379, -0.094923044231589634, 0.00026588909859212508, -0.053474854503345978, 0.069073999037139538]
    accelerations: [-1.9368842023433876e-16, 1.9412643739920391e-12, -7.9423419136843237e-12, 5.4071747403431225e-12, -1.5096876196142427e-14, 3.0404277638322803e-12, -3.935714073298928e-12]
    time_from_start: 1.2
  - positions: [4.1433780730854609e-06, -0.041553916538222824, 0.17018370469438809, -0.11569401509848823, 0.0003240706999660623, -0.065176171649167142, 0.084188706253652712]
    velocities: [3.3995022107659228e-06, -0.034093589492902997, 0.13962999999999623, -0.0949230442315913, 0.00026588909859212974, -0.053474854503346915, 0.069073999037140746]
    accelerations: [-1.935448611626285e-16, 1.9435400354994064e-12, -7.9495171376817828e-12, 5.406840700261507e-12, -1.506974181660724e-14, 3.0395212585253877e-12, -3.9455324028935321e-12]
    time_from_start: 1.3
```

aalmrad, 2025-05-05:
@Abhishek8857 After quickly going over your code, i can see that you are registering the waypoints generated by the moveit planner from the current robot position to a target position published on the "published_coordinates" topic. While I do not have visibility on the publishing node, I assume that the target positions are close to the current robot position and as such your registered waypoints are way too small to be considered as anything but noise. You should only save meaningful joint changes to replay the registered trajectory. I do not know your final objective here but I recommended recording actual joint feedback from the robot instead of the planner's output.

Abhishek8857, 2025-05-06:
@aalmrad Thanks for your reply and a proposed solution.

 Here is a context for the published_coordinated topic. The robot always starts at Zero position where all joint coordinates are at  [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]. The published_coordinated topic publishes joint positions for the robot to reach, one such example of published_coordinates is  home_pose_coordinates = [0.0, 0.0, -0.7650, -3.15, -2.13, 0.006, -1.2, 1.55], where the first value is a flag variable for the code to difeerentiate between the grippper joints or arm joints.