Kinovarobotics/ros2_kortex #283: robootiq_2f_85 gripper error

jamesbeckett35, 2025-05-09:
When I run:
ros2 launch kortex_description view_robot.launch.py gripper:=robotiq_2f_85

I get the following error:
[INFO] [launch]: All log files can be found below /home/james/.ros/log/2025-05-09-08-45-00-331201-jblinux-805971
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command failed. Command: /opt/ros/humble/bin/xacro /opt/ros/humble/share/kortex_description/robots/kinova.urdf.xacro robot_ip:=xxx.yyy.zzz.www name:=kinova arm:=gen3 gripper:=robotiq_2f_85 dof:=7 
Captured stderr output: error: Invalid parameter "isaac_joint_commands"
when instantiating macro: robotiq_gripper (/opt/ros/humble/share/robotiq_description/urdf/robotiq_2f_85_macro.urdf.xacro)
instantiated from: load_gripper (/opt/ros/humble/share/kortex_description/grippers/robotiq_2f_85/urdf/robotiq_2f_85_macro.xacro)
instantiated from: load_robot (/opt/ros/humble/share/kortex_description/robots/kortex_robot.xacro)
in file: /opt/ros/humble/share/kortex_description/robots/kinova.urdf.xacro

aalmrad, 2025-05-09:
Hello,
I was not able to replicate this error on my end. Can you please double check if the locally cloned repository has been modified on your end?

ll16mojh, 2025-05-21:
I've got this exact same issue when I run:

```ros2 launch kortex_bringup kortex_sim_control.launch.py use_sim_time:=true launch_rviz:=false gripper:=robotiq_2f_85```. 

If I don't include the gripper term it seems to work, it opens with the arm without a gripper, and i can control the arm by publishing on the joint position topic. Similarly, when I run 

```ros2 launch kortex_bringup gen3.launch.py   robot_ip:=yyy.yyy.yyy.yyy   use_fake_hardware:=true``` 

rviz opens and I can see the arm, but if i add ```gripper:=robotiq_2f_85``` it gives me the same error:

```Captured stderr output: error: Invalid parameter "isaac_joint_commands"
when instantiating macro: robotiq_gripper (/opt/ros/humble/share/robotiq_description/urdf/robotiq_2f_85_macro.urdf.xacro)
instantiated from: load_gripper (/opt/ros/humble/share/kortex_description/grippers/robotiq_2f_85/urdf/robotiq_2f_85_macro.xacro)
instantiated from: load_robot (/opt/ros/humble/share/kortex_description/robots/kortex_robot.xacro)
in file: /opt/ros/humble/share/kortex_description/robots/gen3.xacro
``` 
 
I'm using Ubuntu 22, ROS2 Humble, aside from the installation instructions for [ROS2](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debs.html), [Gazebo](https://gazebosim.org/docs/latest/ros_installation/) and the Kortex instructions on the [Humble-Gazebo-Classic-Support page](https://github.com/Kinovarobotics/ros2_kortex/tree/Humble_Gazebo_Classic_Support), the only additional things I've done are [enable multicast](https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#enable-multicast) (gazebo doesn't work if I don't do this, just a black window) and ```sudo apt-get install ros-humble-ign-ros2-control``` (because its needed but for some reason isn't installed by default). I have not modified anything, all I've done is installed it and attempted to verify by running the rviz and gazebo simulation commands.

aalmrad, 2025-05-26:
@ll16mojh can you please double check the origin URL of the cloned ros2_robotiq_gripper repository?