Kinovarobotics/ros2_kortex #284: Binary not available as debian in jazzy

ijnek, 2025-05-11:
The package binary is not available on ROS 2 Jazzy.

```bash
$ rosdep resolve kortex_description
#apt
ros-jazzy-kortex-description

$ apt install ros-jazzy-kortex-description
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-jazzy-kortex-description
```

On ROS 2 Rolling, this works:
```
$ apt install ros-rolling-kortex-description
```

aalmrad, 2025-05-12:
@ijnek binary packages are only available in ros2 humble on the Humble_Gazebo_Classic_Support branch

ijnek, 2025-05-15:
I'm surprised this is intentional, becuase the binaries are available in rolling, and kortex_description is listed in the binary releases of [jazzy's distribution.yaml](https://github.com/ros/rosdistro/blob/b77d1cd4511077a2cfe1c29dcc8b8d6e3d62781d/jazzy/distribution.yaml#L7514).

Any chance we can get a release for jazzy so users can perform a binary installation?

ruffsl, 2025-06-04:
Any update on this? I too find it peculiar that binaries are available for Rolling, but the current LTS with Jazzy. It's been so long that another ROS distro has even been released in the meantime with ROS Kilted.

Abban-Fahim, 2025-09-09:
Commenting to bring attention to this issue again. Is there any particular reason for why they haven't been released? Any tasks/issues that need to be finished before they can be released. I could contribute if needed.

aalmrad, 2026-04-03:
As part of Kinova’s beta testing: Claude is analyzing this issue and will attempt an automated fix. A pull request will be opened if changes are made.

130s, 2026-09-15:
Suggest to close. I can at least see apt package is available on Jazzy after https://github.com/ros/rosdistro/pull/49323
```
# apt-cache policy ros-jazzy-kortex-description
ros-jazzy-kortex-description:
  Installed: 0.2.6-1noble.20260905.084346
  Candidate: 0.2.6-1noble.20260905.084346
  Version table:
 *** 0.2.6-1noble.20260905.084346 500
        500 http://packages.ros.org/ros2/ubuntu noble/main amd64 Packages
        100 /var/lib/dpkg/status

# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.4 LTS
```

ijnek, 2026-09-21:
Thanks for the update @130s  , closing.

closed 2026-09-21 by ijnek