Kinovarobotics/ros2_kortex #288: controlling gen3_lite using moveit on ros2 jazzy

perseusdg, 2025-05-20:
Hello,

I have been trying to control a gen3_lite arm with ros2 jazzy both in a gazebo simulation and in a real life robot using moveit. I get the following errors on the gazebo sim : 

```
[rviz2-2] [ERROR] [1747713184.468812464] [rviz2.moveit.ros.move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
[rviz2-2] [INFO] [1747713184.736282133] [rviz2.moveit.ros.move_group_interface]: MoveGroup action client/server ready
[rviz2-2] [INFO] [1747713184.736609647] [rviz2.moveit.ros.move_group_interface]: Planning request accepted
[rviz2-2] [INFO] [1747713184.744712050] [rviz2.moveit.ros.move_group_interface]: Planning request aborted
[rviz2-2] [ERROR] [1747713184.744864695] [rviz2.moveit.ros.move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
[rviz2-2] [INFO] [1747713184.969518311] [rviz2.moveit.ros.move_group_interface]: MoveGroup action client/server ready
[rviz2-2] [INFO] [1747713184.969823977] [rviz2.moveit.ros.move_group_interface]: Planning request accepted
[rviz2-2] [INFO] [1747713184.989156680] [rviz2.moveit.ros.move_group_interface]: Planning request aborted

```

while in the physical arm i get the following errors : 
```
rviz2-1] [ERROR] [1747677141.472947616] [rviz2_moveit.moveit.ros.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1747677141.488677686] [rviz2_moveit.moveit.ros.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-1] [INFO] [1747677141.559106468] [rviz2_moveit.moveit.ros.rdf_loader]: Loaded robot model in 0.00585045 seconds
[rviz2-1] [INFO] [1747677141.559224695] [rviz2_moveit.moveit.core.robot_model]: Loading robot model 'gen3_lite_gen3_lite_2f'...
[rviz2-1] [INFO] [1747677141.559268766] [rviz2_moveit.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-1] [WARN] [1747677141.704470387] [rviz2_moveit.moveit.core.robot_model]: Could not identify parent group for end-effector 'end_effector'
[rviz2-1] [INFO] [1747677141.708666063] [rviz2_moveit.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'arm': 1 1 1 1 1 1
[rviz2-1] [INFO] [1747677142.264528386] [rviz2_moveit.moveit.ros.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1747677142.265032086] [rviz2_moveit.moveit.ros.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-1] [INFO] [1747677142.460111260] [interactive_marker_display_105789966460944]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-1] [INFO] [1747677142.468229514] [rviz2_moveit.moveit.ros.motion_planning_frame]: group arm
[rviz2-1] [INFO] [1747677142.468250362] [rviz2_moveit.moveit.ros.motion_planning_frame]: Constructing new MoveGroup connection for group 'arm' in namespace ''
[rviz2-1] [INFO] [1747677142.474275383] [rviz2_moveit.moveit.ros.move_group_interface]: Ready to take commands for planning group arm.
[rviz2-1] [INFO] [1747677142.474977797] [rviz2_moveit.moveit.ros.motion_planning_frame]: group arm
[rviz2-1] [INFO] [1747677142.559986548] [interactive_marker_display_105789966460944]: Sending request for interactive markers
[rviz2-1] [INFO] [1747677142.635680970] [interactive_marker_display_105789966460944]: Service response received for initialization
[rviz2-1] [INFO] [1747677152.757800185] [rviz2_moveit.moveit.ros.move_group_interface]: MoveGroup action client/server ready
[move_group-2] [INFO] [1747677152.758320501] [move_group.moveit.moveit.ros.move_group.move_action]: MoveGroupMoveAction: Received request
[rviz2-1] [INFO] [1747677152.758558832] [rviz2_moveit.moveit.ros.move_group_interface]: Planning request accepted
[move_group-2] [INFO] [1747677152.758646999] [move_group.moveit.moveit.ros.move_group.move_action]: executing..
[move_group-2] [INFO] [1747677152.759810045] [move_group.moveit.moveit.ros.move_group.move_action]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-2] [INFO] [1747677152.759855259] [move_group.moveit.moveit.ros.move_group.capability]: Using planning pipeline 'ompl'
[move_group-2] [INFO] [1747677152.760092675] [move_group]: Calling PlanningRequestAdapter 'ResolveConstraintFrames'
[move_group-2] [INFO] [1747677152.760145390] [move_group]: Calling PlanningRequestAdapter 'ValidateWorkspaceBounds'
[move_group-2] [INFO] [1747677152.760166797] [move_group]: Calling PlanningRequestAdapter 'CheckStartStateBounds'
[move_group-2] [ERROR] [1747677152.760215797] [move_group.moveit.moveit.ros.check_start_state_bounds]: Joint 'joint_3' from the starting state is outside bounds by: [2.79816 ] should be in the range [-2.61 ], [2.61 ].
[move_group-2] [ERROR] [1747677152.760262544] [move_group]: PlanningRequestAdapter 'CheckStartStateBounds' failed, because 'Start state out of bounds.'. Aborting planning pipeline.
[move_group-2] [ERROR] [1747677152.760285265] [move_group.moveit.moveit.ros.move_group.move_action]: Generating a plan with planning pipeline failed.
[move_group-2] [INFO] [1747677152.760340352] [move_group.moveit.moveit.ros.move_group.move_action]: FAILURE
[rviz2-1] [INFO] [1747677152.760565471] [rviz2_moveit.moveit.ros.move_group_interface]: Planning request aborted
[rviz2-1] [ERROR] [1747677152.761237334] [rviz2_moveit.moveit.ros.move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
```

I tried to launch other arms in simulation like the gen3 with robotiq_2f_85 and I was able to successfully control it using moveit. So i decided to dig a bit deeper and try controlling the gen3_lite arm with the kortex library for ros-noetic (and moveit 1) and I was able to successfully control it - upon comparing the config files across both kortex1 and kortex2 for gen3_lite - i noticed that in kortex2 - none of the joints have acceleration limits - https://github.com/Kinovarobotics/ros2_kortex/blob/b7d58a62d0230693a0e820fbf5a1a59aaaf2c3a9/kortex_moveit_config/kinova_gen3_lite_moveit_config/config/joint_limits.yaml#L14  while kortex1 has acceleration limits defined in it for the same arm - https://github.com/Kinovarobotics/ros_kortex/blob/bf71d8ecdec874a88d02b1a2481784ae2a7fdba4/kortex_move_it_config/gen3_lite_gen3_lite_2f_move_it_config/config/default_joint_limits.yaml#L13 -  could the absence of acceleration limits be the root issue behind why gen3_lite isn't working with ros2 jazzy for me ?  or is the issue something else ?
any tips on this issue will be immensely helpful ! (I would prefer to not use ros1 bridge since the rest of my codebase works with ros2 jazzy)

aalmrad, 2025-05-22:
Hello @perseusdg , the control of of gen3 lite with moveit will be investigated in both real-life and simulation. I will report back soon on this.

Best,
Abed

aalmrad, 2025-06-04:
@perseusdg, can you please create a pull request so i can review the modifications and possibly merge?
Only one joint in the gripper is supposed to move and the rest should mimic its motion, i'll look into this once you create the pull request. Thanks for your contribution

perseusdg, 2025-07-08:
I will create a pull request for this on the weekend , got a bit busy with school work !  I think I know why the gripper isn't planning proper in the sim , I will see if I can push a fix for that too