Kinovarobotics/ros2_kortex #289: Terrible branch naming convention and development practices

fzoric8, 2025-05-20:
Hi, 

I'm writing to you because I spent a day debugging unnecessary and unwanted merge caused by the bad naming conventions and lack of systematization in the driver development. I can accept part of the responsibility for that merge, but it wouldn't happen if branch names were in place. 

Branch names are total chaos, and there is no clear reference to which ROS 2 distro is something related. 

I would urge you to clean that mess up - to enable easy and simple usage for the `ros2_kortex` driver. 

I don't understand why would you move `humble` branch and why should it be related to the gazebo classic only.

aalmrad, 2025-05-22:
Hello @fzoric8,

Thanks for sharing your experience and sorry to hear about your confusion with the branches names.

Just to clarify:

The Humble_Gazebo_Classic_Support branch was originally named to highlight that it retains Gazebo Classic support for ROS 2 Humble. However, it’s not limited to Gazebo Classic. It is also compatible with Ignition Fortress, and it’s currently the most stable and up-to-date branch for ROS 2 Humble users. We’ll look into either renaming the branch (e.g., humble_stable) or updating the README to clarify this.

Regarding the systematization:

The ros2_kortex repository has grown organically over time to support multiple ROS 2 distributions and simulators (Gazebo Classic, Ignition), and we recognize that this has introduced some inconsistencies in naming, structure, and maintenance.

We're currently taking steps to improve this, including:

Clarifying the purpose and compatibility of each branch (via README tables and renaming where appropriate).

Reviewing the use of consistent naming conventions across drivers, plugins, and controller interfaces.

Reinforcing the branching strategy (e.g., stable vs development, feature branches).

Adding distro-specific notes or CMake flags to avoid ambiguity during builds.

If you have concrete suggestions we are open for your input. Thanks again for taking the time to raise this — your feedback is valuable.

Best,
Abed

fzoric8, 2025-06-04:
Hi @aalmrad, 

Thank you for your response. 

I suggest that there is a branch name that corresponds to the ROS distro that is inteded for usage. 

+ I would also suggest decoupling `humble` branch and `humble_gazebo_classic`. 

Best, 
Filip

closed 2025-06-04 by fzoric8