Kinovarobotics/ros2_kortex #294: `gripper_controllers` dependency deprecated in Kilted and later

sea-bass, 2025-06-29:
The `gripper_controllers` package was replaced with a `parallel_grippers_controller` package in Kilted and later (current Rolling).

See https://github.com/moveit/moveit2_tutorials/pull/1045#pullrequestreview-2969578999 for information on how to fix this.

christophfroehlich, 2025-09-03:
@aalmrad In the meantime, we removed the gripper_controllers from rolling and kilted.
https://github.com/Kinovarobotics/ros2_kortex/blob/309f9c9d4a277970e542e5ac1fe260ced0630f65/kortex_bringup/package.xml#L25

aalmrad, 2026-04-03:
As part of Kinova’s beta testing: Claude is analyzing this issue and will attempt an automated fix. A pull request will be opened if changes are made.